ethz-asl/rotors_simulator #480: How to interface rotorS with an custom made quadrotor

chrismath, 2018-04-11:
I am currently working on quadrotors to move it from one point to the other.So I wanted to implement this on an hardware and the see it in real time.Could anybody help so that I can interface it with pix hawk controller.

nkhedekar, 2018-06-07:
You can use mavros and px4 with Gazebo as is given here : https://github.com/nkhedekar/offb_py

or you can simply publish the waypoints by using `rostopic pub` after spawning a multirotor available in RotorS. 

This can be done by-
` $ roslaunch rotors_gazebo mav_hovering_example.launch`
and then
` $ rostopic pub /firefly/command/pose geometry_msgs/PoseStamped "pose:
  position:
    x: 3.0
    y: 4.0
    z: 5.0
  orientation:
    x: 0.0
    y: 0.0
    z: 0.0
    w: 0.0" 
`

*The values of the variables being the point you want the multirotor to move to

nkhedekar, 2018-06-07:
This might also be helpful for the implementation on hardware : https://dev.px4.io/en/ros/offboard_control.html