ethz-asl/rotors_simulator #487: Controlling actuator velocities using simulink blocks

HusseinHazem, 2018-05-02:
I launched the mav_hovering_example.launch file and commented out lines 32 till line 36 (the part where the lee_position_controller node launches) to be able to control the speeds of the motors using the message publisher in rqt. 
Is it possible to control each motor speed from the simulink ros blocks? and if so how can i do it? 
My attempt leads to gazebo freezing and even then i don't how to access each motor speed separately.

gsilano, 2018-05-02:
I suggest you to read something about the MathWorks [Robotics System Toolbox](https://it.mathworks.com/products/robotics.html?requestedDomain=)

HusseinHazem, 2018-05-02:
![screenshot from 2018-05-02 15 58 33](https://user-images.githubusercontent.com/36478570/39527571-0b638eb2-4e22-11e8-88fd-4ec5c4b672b5.png
that was my attempt.
but i also followed https://it.mathworks.com/help/robotics/examples/work-with-ros-messages-in-simulink.html but i couldn't get it to work.

gsilano, 2018-05-02:
I suggest you to read these tutorials: [Connect to a ROS-enabled Robot from Simulink](https://it.mathworks.com/help/robotics/examples/connect-to-a-ros-enabled-robot-from-simulink.html), [Install robotics system toolbox support packages](https://it.mathworks.com/help/robotics/ug/install-robotics-system-toolbox-support-packages.html) and [Create custom messagges from ROS package](https://it.mathworks.com/help/robotics/ug/create-custom-messages-from-ros-package.html). 

You have to take in mind that _mav_msgs/Actuators_ is not a standard ROS message. It is not present in the default MATLAB installation (with the Robotics System Toolbox) and you have to import it by using the tutorials I linked you.

HusseinHazem, 2018-05-02:
thanks but i already went through all of these and installed ROS custom message interface. and as you can see from the screenshot the message type mav_msgs/Actuators in available for use. My problem is how to publish the message to the motor speeds.

gsilano, 2018-05-02:
You right!! It is not allowed to publish a message via Simulink, at least I did not succeed. You should use MATLAB, in particular the commands as reported below:

[pub, msg] = rospublisher('/command/motor_speed','mav_msgs/Actuators');
msg.AngularVelocities = [Omega_1_SIMULINK.signals.values(end), Omega_2_SIMULINK.signals.values(end), Omega_3_SIMULINK.signals.values(end), Omega_4_SIMULINK.signals.values(end)];
send(pub,msg);

where Omega_1, Omega_2, and so on, are the propellers angular velocities you would publish. 

I hope I have made it clear.

HusseinHazem, 2018-05-03:
[image]

Thanks a lot , i changed it a bit to the above and it worked. My goal is to use my controller from simulink to control the motor velocities, any idea how i could do that?

gsilano, 2018-05-03:
Great!! You have to combine SIMULINK with MATLAB. The MATLAB script handles the communication, receiving data from ROS and passing such data to Simulink. The control scheme in Simulink computes the control actions and saves such actions in a toWorkspace block. Later MATLAB, thanks to the command I showed (and you corrected a little bit), sends the motor command to GAZEBO.

To do that you have to synchronize the enviroment by using the GAZEBO services pause and unpase physics.

HusseinHazem, 2018-05-05:
Do you know how i can remove this error? the u here is a vector of length 4 coming from a mux and for some reason the function can't recognize the ros function.
[image]
[image]

gsilano, 2018-05-05:
You cannot use this kind of Matlab function in Simulink. I mean, you cannot publish via Simulink by using it. Tomorrow morning I will tell (editing this message) how to do 😊

HusseinHazem, 2018-05-06:
thanks i will try it out and get back to you, but will 2017a version not work?

gsilano, 2018-05-06:
I made the Simulink scheme by using 2018a version of MATLAB. I don't know if you can open it with a previous version of the software. You should try.

HusseinHazem, 2018-05-06:
it is possible but you have to change the file from a simulink 2018a as shown in the pic
[image]
Can you export to version 2017a and send it again?

gsilano, 2018-05-06:
Yes, of course. You find it in attach.

[GitHub_issue_RotorS_MALTAB2017a.zip](https://github.com/ethz-asl/rotors_simulator/files/1977717/GitHub_issue_RotorS_MALTAB2017a.zip)

HusseinHazem, 2018-05-06:
Thanks a lot, but i am having a bit of trouble understanding how i am supposed to run it using my control method from simulink. Am i supposed to send my omegas generated from my control to the omega blocks instead of the constants and run the simulink model first then the matlab script?

gsilano, 2018-05-06:
As I told you before, the MATLAB script has the aim of handling the communication. It is enough to run the MATLAB script to execute the entire simulation. 
Thus, you have to think in your mind an event triggered system. Such system brings data from the environment, computes the next action to do (the propeller angular velocities) and acts it in order to reach the goal. Every step, each Ts seconds (1e-3 in my example) the cycle is run. How many time? The number of the simulation steps.

In other words, the code works like an automata. In each state, some actions are done and later the results used by the next node, and so on.

HusseinHazem, 2018-05-07:
so that being said, i won't be able to use my MPC (model predictive control) i did on simulink where i publish the speed of the motors depending on the feedback i get from gazebo (i.e position of the hummingbird)

HusseinHazem, 2018-05-07:
i hadn't noticed that you run the simulink simulation in the matlab script, so it should work?

gsilano, 2018-05-07:
Yes, you are able to do that..but you have to redesign the controller. I mean, the idea behind. 
I already used the Robotics System Toolbox in my research activity. In particular, I employed such toolbox to develop an extension of the RotorS platform named [CrazyS](https://github.com/gsilano/CrazyS).

In this [video](https://youtu.be/ZPyMnu7A11s) is showed as the Robotics System Toolbox (on a Windows machine) is able to send commands (the propellers angular velocities) to and receive data from the Gazebo 3D virtual environment (on a Ubuntu 14.04 machine).

HusseinHazem, 2018-05-07:
So if this is my controller with the system, i should just change the feedback to be of x y z roll pitch and yaw to be from the workspace of matlab which come from gazebo?

[image]

gsilano, 2018-05-07:
First of all, when you are using the Robotics System Toolbox with Gazebo you have to take in mind that your model is in Gazebo, and not in MATLAB. Later, the feedback loop data come from Gazebo. Thus, you have to read the odometry messages from the environment, to compute from the propellers angular velocities for a single step, and finally to send them by using the properly MATLAB function.

HusseinHazem, 2018-05-07:
Yes i know, i will totally neglect the system in matlab
[image]
should that work?

gsilano, 2018-05-07:
Yes, it should work.

HusseinHazem, 2018-05-07:
The "from workspace " blocks give me that error
[image]

gsilano, 2018-05-07:
You have to save the variables in the Workspace as 2D Array. See my Simulink scheme.

HusseinHazem, 2018-05-07:
It worked, flies for like 3 seconds then it disappears into the ground hahahahahah but i think it is an error from my control.
[image]

gsilano, 2018-05-07:
Maybe, could be. I am glad to listen that all is ok now. Thus, I think you can close the issue now.

Say something more in the title. Such discussion could help others that are interested to use the toolbox but don't know how to do.

HusseinHazem, 2018-05-07:
i don't think it is the control though, are there any parameters that i can change which would effect the simulation somehow?

gsilano, 2018-05-07:
The simulation steps, maybe. It has to be equal to the simulation step that you used in the simulation scheme. Search on google how to modify a world in GAZEBO, the real time factor.

gsilano, 2018-05-07:
Why not. It is the minimum for the help you gave me some months ago: the issues I had with the build when I opened the pull request #465.

Could I insert such guide in [CrazyS](https://github.com/gsilano/CrazyS)? If you remember, it is an extension of the RotorS capabilities in which I modeled, developed and integrated the Crazyflie 2.0 nano-quadcopter.

ffurrer, 2018-05-08:
This is up to you of course, but I guess it would make a bit more sense if it would be in this repo. As this is unspecific to CrazyS.

gsilano, 2018-05-08:
Yes, of course. I will write it for RotorS sharing such contents also in CrazyS. In this way, the same procedure may be used not only by RotorS developers but also the CrazyS ones. 
I remember well that CrazyS is nothing else than a branch, an extension of RotorS.

HusseinHazem, 2018-05-09:
Giuseppe, is there a way to run the simulink for one step? because when i put my controller scheme it generates the input till the quadrotor settles so for example it can reach z=1 then fall and in the next simulation step it will do the same and so on.

HusseinHazem, 2018-05-09:
what kinda worked for me was to do the whole thing on matlab, i tried out a simple PID for Z. it goes to the desired Z position and keeps hovering for about 40 seconds perfectly then it drifts slowly and eventually falls. Can you give me your email so i can send you?

gsilano, 2018-05-10:
My e-mail address is giuseppe.silano[at]unisannio.it :-)

gsilano, 2018-11-19:
@ffurrer, I just updated the wiki section of RotorS related issue above. Here the link

https://github.com/ethz-asl/rotors_simulator/wiki/Interfacing-RotorS-through-Matlab

In the coming days, I will release a new version of CrazyS with all the files I mentioned in the wiki. I hope it will be useful for all RotorS users.

mahdinobar, 2019-03-27:
@gsilano Hello, I was wondering if you would inform me of the documentation or reference for the hovering example controller model at [https://github.com/ethz-asl/rotors_simulator/wiki/Interfacing-RotorS-through-Matlab#examples-controller](url).

gsilano, 2019-03-27:
Hi @mahdinobar! I don't know if I understand correctly, but all the information regarding the use of the MathWorks Robotics System Toolbox with RotorS are reported in the Wiki section of [CrazyS](https://github.com/gsilano/CrazyS) and RotorS. Are you looking for other stuff?

islamdib, 2020-04-11:
> 
> 
> So if this is my controller with the system, i should just change the feedback to be of x y z roll pitch and yaw to be from the workspace of matlab which come from gazebo?
> 
> [image]

Hello,
please can u send me your model ?

gsilano, 2020-04-11:
Here it is the link to the Wiki page: https://github.com/ethz-asl/rotors_simulator/wiki/Interfacing-RotorS-through-Matlab

danialwaleed, 2022-04-16:
Hi Everyone,

I have been working on a similar problem which [HusseinHazem](https://github.com/HusseinHazem)  was working on and that is how to publish motor speed commands using Simulink. I believe I have figured out a method to do so. I am attaching a simulink file as well and hope it helps. 

The solution requires the use of  "Use of Matlab function to assign nested arrays" documentation in Matlab, I have done this for the mav_msgs/Actuators message.

In the function make sure that the message length is defined using the command

msg.AngularVelocities_SL_Info.CurrentLength=uint32(4);

After that assign values to the motor using the command
msg.AngularVelocities=u;

Where u is the input to the function. I am using Matlab 2021b so if you need an older version do let me know. I have been trying to figure this problem out for quite a few months and I hope this helps other people.

[Simulink_motor_pub.zip](https://github.com/ethz-asl/rotors_simulator/files/8499227/Simulink_motor_pub.zip)