ethz-asl/rotors_simulator #496: I dont know how to publish velocity into MultiDOFJointTrajectory.h

chngdickson, 2018-06-11:
I've been trying to publish trajectory_msg.velocities.linear.x=1; to control the velocity of the drone on the firefly using the message file which can be found by typing

rosmsg show trajectory_msgs/MultiDOFJointTrajectory
 
to publish message into /firefly/command/trajectory.

I've also tried publishing it into mav_msgs::default_topics::COMMAND_TRAJECTORY
but i don't get how to do it. Please guide me or at least show me to a correct documentation where i can learn it and get my program working

Thank you for reading this issue.

helenol, 2018-06-12:
Hi @chngdickson,
This velocity will not be tracked unless you have a velocity controller. The lee controller that ships with rotors is a position controller, so it will attempt to track the position, and if I remember correctly may even ignore lower derivatives.
Please see https://github.com/ethz-asl/rotors_simulator/issues/490 for suggestions, but if you want to directly command velocity, you'll have to write your own controller.

chngdickson, 2018-06-13:
hello @helenol , You've explained the question perfectly. 
I am quite new in ROS and i was wondering if you could still help me with additional issues i have.
There are 2 other things i am still struggling with.

First, I don't quite understand how to subscribe to the Odom topic, i've tried subscribing to 
"/firefly/odometry_sensor1" with message nav_msgs::Odometry as listed in rostopic info...

Second, i am also trying to subscribe to the vi_sensor/depths
to get the distances in each second.

I've tried to work around the mapping using rqt_graph but i am still as clueless as ever.
Thank you in advance for the reply and thank you for helping me answer this question previously.
I also apologize for asking another question without creating another issue in addition to my incompetence. I want to learn, please guide me the way.

Thank you.