ethz-asl/rotors_simulator #510: Publishing waypoints using Python code, adding lidar sensor and the frame_id issue

Amir-Ramezani, 2018-08-05:
Thanks for your package.

I am trying to publish waypoints in python using the following code:

```
    from trajectory_msgs.msg import MultiDOFJointTrajectory
    from geometry_msgs.msg import Vector3
    import mav_msgs.msg as mav_msgs

    Firefly_Command_Publisher = rospy.Publisher('/firefly/command/trajectory', MultiDOFJointTrajectory, queue_size=10)

    trajectory_msg=MultiDOFJointTrajectory()
    trajectory_msg.header.stamp=rospy.Time.now()

    desired_position = Vector3()
    desired_position.x = 1.0
    desired_position.y = 1.0
    desired_position.z = 1.0
    desired_yaw = 45

    mav_msgs.msgMultiDofJointTrajectoryFromPositionYaw(desired_position, desired_yaw,trajectory_msg)
    Firefly_Command_Publisher.publish(trajectory_msg)
```

but I get the following error:

```
    mav_msgs.msgMultiDofJointTrajectoryFromPositionYaw(desired_position, desired_yaw,trajectory_msg)
AttributeError: 'module' object has no attribute 'msgMultiDofJointTrajectoryFromPositionYaw'
```

I don't know how to add `msgMultiDofJointTrajectoryFromPositionYaw`, I used `import mav_msgs.msg as mav_msgs` and `from conversions import msgMultiDofJointTrajectoryFromPositionYaw` but no luck, I appreciate your guide on this. Further, if you know any other way that I can use a controller to control your drone from python using commands such as move forward and turn left or right please tell me.

Beside that, in Gazebo 2.0 I am trying to add a lidar sensor using the following code:

```
 <!-- Hokuyo Lidar Range Sensor -->
    <xacro:macro name="hokuyo_utm30lx" params="*origin update_rate ray_count min_angle max_angle">
      <link name="firefly/laser_lidar_link12">
	<inertial>
	  <mass value="0.001" />
	  <origin xyz="0 0 0" rpy="0 0 0" />
	  <inertia ixx="0.0001" ixy="0" ixz="0" iyy="0.0001" iyz="0" izz="0.0001" />
	</inertial>
	<visual>
          <geometry>
            <mesh filename="package://turtlebot_description/meshes/sensors/hokuyo_utm30lx/hokuyo_utm_30lx.dae"/>
          </geometry>
	</visual>
	<collision>
	  <origin xyz="0 0 -0.0115" rpy="0 0 0" />
          <geometry>
            <box size="0.058 0.058 0.087" />
            <!--<mesh filename="package://hector_sensors_description/meshes/hokuyo_utm30lx/hokuyo_utm_30lx.stl"/>-->
          </geometry>
	</collision>
      </link>
      <joint name="laser_lidar_joint" type="fixed">
        <origin xyz="0.0 0.0 0.12" rpy="3.14159 0.0 0.0" />
	<parent link="firefly/base_link"/>
	<child link="firefly/laser_lidar_link12"/>
      </joint>
      <gazebo reference="firefly/laser_lidar_link12">
        <sensor type="ray" name="firefly_laser_lidar">
            <pose>0 0 0 0 0 0</pose>
            <visualize>false</visualize>
            <update_rate>40</update_rate>
	    <ray>
		  <scan>
		    <horizontal>
		      <samples>720</samples>
		      <resolution>1</resolution>
		      <min_angle>-2.350796</min_angle>
		      <max_angle>2.350796</max_angle>
		    </horizontal>
		  </scan>
		  <range>
		    <min>0.10</min>
		    <max>20.0</max>
		    <resolution>0.01</resolution>
		  </range>
		  <noise>
		    <type>Gaussian</type>
		    <mean>0.0</mean>
		    <stddev>0.01</stddev>
		  </noise>
	    </ray>
	    <plugin name="gazebo_ros_lidar_controller" filename="libgazebo_ros_laser.so">
		  <topicName>/scan</topicName>
		  <frameName>firefly/laser_lidar_link12</frameName>
		  <!-- <interface:laser name="gazebo_ros_${laser_suffix}_iface" /> -->
            </plugin>
        </sensor>
      </gazebo>
    </xacro:macro>
```

but the output of `rostopic echo /scan/header` is:

```
seq: 721
stamp: 
  secs: 6850
  nsecs: 560000000
frame_id: firefly/firefly/laser_lidar_link12
---
seq: 722
stamp: 
  secs: 6850
  nsecs: 590000000
frame_id: firefly/firefly/laser_lidar_link12
```

my problem is that the frame id is changed from what I have defined using xacro macro - I expected it to be /firefly/laser_lidar_link12, and as a result, I cannot use it correctly in some SLAM packages - as I explained it in this link https://github.com/introlab/rtabmap_ros/issues/266, is there any way to solve this issue?

chrismath, 2018-08-22:
I thinks its the way your publishing to the topic i have done it like this
traj = MultiDOFJointTrajectory()
		traj.header.frame_id =''
		traj.header.stamp = rospy.Time.now()
		traj.joint_names = ["base_link"]

		transforms =Transform()
		transforms.translation.x=ax
		transforms.translation.y=ay
		transforms.translation.z=1

		velocities =Twist()
		accelerations=Twist()
		point = MultiDOFJointTrajectoryPoint([transforms],[velocities],[accelerations],rospy.Duration(0.1))
		traj.points.append(point)
		self.trajectory_pub.publish(traj)

Amir-Ramezani, 2018-08-22:
Thanks chrismath, I test it but it doesn't work. I run the code successfully but it doen't move the firefly.

nkhedekar, 2018-08-23:
Are you able to get it to move by publishing positions on the `command/pose` topic? If yes, and if sending a trajectory is not required, you could just publish the points iteratively.

Amir-Ramezani, 2018-08-23:
I can move the rotor using the following command:
`rosrun rotors_gazebo waypoint_publisher 1 1 1.5 0 0 __ns:=firefly
`

and also by publishing to `commnd/pose` topic using rqt

But cannot move it directly by the following python code:

```
from trajectory_msgs.msg import MultiDOFJointTrajectory, MultiDOFJointTrajectoryPoint
from geometry_msgs.msg import Twist 
from geometry_msgs.msg import Transform

Firefly_Command_Publisher = rospy.Publisher('/firefly/command/trajectory', MultiDOFJointTrajectory, queue_size=10)

traj = MultiDOFJointTrajectory()
traj.header.frame_id =''
traj.header.stamp = rospy.Time.now()
traj.joint_names = ["base_link"]

transforms =Transform()
transforms.translation.x=1.0
transforms.translation.y=1.0
transforms.translation.z=2.0
velocities =Twist()
accelerations=Twist()
point = MultiDOFJointTrajectoryPoint([transforms],[velocities],[accelerations],rospy.Duration(0.1))
traj.points.append(point)
Firefly_Command_Publisher.publish(traj)

```

chrismath, 2018-08-23:
Amir-Ramezani , please upause the gazebo,as suggested by nkhedekar  .

Amir-Ramezani, 2018-08-23:
Chrismath, did you read my previous comment? I mentioned that I can move the rotor using rqt and by publishing command in terminal, it means gazebo is un-paused.

chrismath, 2018-08-24:
ok, then i dont know Amir-Ramezani ,the above code that i mentioned is working for me

nkhedekar, 2018-08-27:
Please try the following code:
```
from trajectory_msgs.msg import MultiDOFJointTrajectory, MultiDOFJointTrajectoryPoint
from geometry_msgs.msg import Twist 
from geometry_msgs.msg import Transform

firefly_command_publisher = rospy.Publisher('/firefly/command/trajectory', MultiDOFJointTrajectory, queue_size=10)

traj = MultiDOFJointTrajectory()
transforms =Transform()
velocities =Twist()
accelerations=Twist()
point = MultiDOFJointTrajectoryPoint([transforms],[velocities],[accelerations],rospy.Duration(1))
traj.points.append(point)
transforms =Transform()
transforms.translation.x=1.0
transforms.translation.y=1.0
transforms.translation.z=2.0
velocities =Twist()
accelerations=Twist()
point = MultiDOFJointTrajectoryPoint([transforms],[velocities],[accelerations],rospy.Duration(2))
traj.points.append(point)

firefly_command_publisher.publish(traj)
```
This can be run with the mav_hovering_example. 

Apart from this, you can also get this repository: https://github.com/ethz-asl/mav_control_rw  and use the controllers available in it. To run the simulator and the linear MPC, in seperate terminals run the following commands:
```
  $ roslaunch rotors_gazebo mav.launch mav_name:=firefly

  $ roslaunch mav_linear_mpc mav_linear_mpc_sim.launch mav_name:=firefly
```

CarSanoja, 2019-01-04:
Hello,
It is not clear to me how the package transforms that point (.../command/pose x:0 y:0 z:1 for example) into a trajectory. Where it reflects the behavior that must have in the period between the initial and the  final point.

@Amir-Ramezani Your work looks impressive, can you share me a little more about it? It calls my attention to know how you modeled the policies to indicate to the drone when it suffered or had a collision or a success action.

I hope you can help me,
Best regards

Translated with www.DeepL.com/Translator

juanmed, 2019-05-28:
For anyone looking at this, the following also works for me.

```python
#!/usr/bin/env python

# Waypoint publisher for RotorS
# Reference: https://github.com/ethz-asl/rotors_simulator/issues/510
# 

import rospy
import sys
import tf
import numpy as np

from trajectory_msgs.msg import MultiDOFJointTrajectory, MultiDOFJointTrajectoryPoint
from geometry_msgs.msg import Twist 
from geometry_msgs.msg import Transform

def publish_waypoint(x,y,z,yaw):
	"""
	Publish a waypoint to 
	"""

	command_publisher = rospy.Publisher('/firefly/command/trajectory', MultiDOFJointTrajectory, queue_size = 10)

	# create trajectory msg
	traj = MultiDOFJointTrajectory()
	traj.header.stamp = rospy.Time.now()
	traj.header.frame_id = 'frame'
	traj.joint_names.append('base_link')

	# create start point for trajectory
	transforms = Transform()
	velocities = Twist()
	accel = Twist()
	point = MultiDOFJointTrajectoryPoint([transforms],[velocities],[accel],rospy.Time(1))
	traj.points.append(point)

	# create end point for trajectory
	transforms = Transform()
	transforms.translation.x = x
	transforms.translation.y = y
	transforms.translation.z = z 

	quat = tf.transformations.quaternion_from_euler(yaw*np.pi/180.0, 0, 0, axes = 'rzyx')
	transforms.rotation.x = quat[0]
	transforms.rotation.y = quat[1]
	transforms.rotation.z = quat[2]
	transforms.rotation.w = quat[3]

	velocities = Twist()
	accel = Twist()
	point = MultiDOFJointTrajectoryPoint([transforms],[velocities],[accel],rospy.Time(2))
	traj.points.append(point)

	rospy.sleep(1)
	command_publisher.publish(traj)

if __name__ == '__main__':
	try:

		rospy.init_node("riseq_rotors_waypoint_publisher", anonymous = True)

		# get command line params
		x_des = float(sys.argv[1])
		y_des = float(sys.argv[2])
		z_des = float(sys.argv[3])
		yaw_des = float(sys.argv[4])

		publish_waypoint(x_des, y_des, z_des, yaw_des)

		#rospy.spinOnce()
		rospy.loginfo(" >> Published waypoint: x: {}, y: {}, z: {}, yaw: {}".format(x_des, y_des, z_des, yaw_des))

	except rospy.ROSInterruptException:
		print("ROS Terminated")
		pass
```

First start rotors 

```bash
roslaunch rotors_gazebo mav_hovering_example.launch
```
then call the script indicating, for example,  the desired position as (1,1,1) and yaw = 0 degrees :
```bash
rosrun riseq_tests waypoint_publisher.py 1 1 1 0 __ns:=firefly
```

IhabMohamed, 2019-06-20:
Hi everyone,

I would like to know how to write a trajectory Callback function in order to subscribe to "hummingbird/command/trajectory" using python.

I have tried many ways, but this is the one works. However, I can not extract, for example, the position.

def trajectoryCallback(traj_msg):
    points = traj_msg.points

I really appreciate your help

Regards,
Ihab

harshkakashaniya, 2019-10-27:
@Amir-Ramezani Awesome work.
But in your code drone goes to the origin and reaches the waypoint.
Can we directly go from a point to another waypoint 
Thank you in advance..

Amir-Ramezani, 2019-10-29:
@harshkakashaniya 
Thanks, yes, it is possible, check this paper https://arxiv.org/pdf/1707.05110.pdf. They also trained the drone using RL but with the help of PD controller. In our work we also used RL (a different RL method) to train but with no PD or another conventional controller so our method is quiet different comparing to that paper. That being said, the idea is that you always train the drone to go to one point (the best is origin, in our case we need to have the height to avoid hitting ground, so (0,0,1.5) = Point A), and if you want the drone to move from point A to point B, you should subtract Point B from point A. For example, if you want to go to point B = (1,0,1.5) from point A, then you just tell the drone that you are in point (-1,0,1.5), so it tries to go to point (0, 0, 1.5) = Point A. So the drone moves X=+1 in order to be in (1,0,1.5). In another word and as mentioned in that paper "During policy optimization, we train it to go to the origin of the inertial frame. During operation, we input the state subtracted by the waypoint location to the policy. This way we do not have to train waypoint tracking explicitly."

juanmed, 2019-10-29:
@Amir-Ramezani I have been wondering for a long time about your paper related to this topic. Could you please share a link to it? It would like to study your contribution. Thanks.

harshkakashaniya, 2019-10-29:
@Amir-Ramezani 
Thanks for the intuition of transformation I will go through the paper.