ethz-asl/rotors_simulator #514: control parameters

elisabethwelburn12, 2018-08-23:
Hi,

I have input my own drone model using these files -e.g. made a new xacro file and changed the mass and inertial files but still calling the original lee_position_controller and waypoint_publisher files. I know I have to tune the parameters set out in the lee_controller_{mav_name}.yaml file that include the following parameters: 
```
position_gain: {x: 4, y: 4, z: 4}
velocity_gain: {x: 2.7, y: 2.7, z: 2.7}
attitude_gain: {x: 10, y: 10, z: 0.35}
angular_rate_gain: {x: 0.22, y: 0.22, z: 0.01}

The issue is I'm not sure what each of these is exactly doing as it isn't a standard PID controller. If somebody could give some guidance on their significance that would be great.

Thanks
```

rishikeshrmadan, 2019-09-26:
Hi @elisabethwelburn12 !
I'm struggling with this as well, did you have any luck? I went through the resources of the controller, but can't break through...

elisabethwelburn12, 2019-09-26:
Hi, to save you buckets of time try the hector_quadcopter ROS package. I reappropriated it and its the only drone simulation that I got working into the end using ROS everything else seems to be out of date or broken somehow. Best of luck !

rishikeshrmadan, 2019-09-27:
Thanks a ton! I've literally sunk like a 100 hours into this already, in the overall drone simulation I mean, the model, it's parameters etc.. I'm fully aware of the fallacy of sunk costs(!) and since I've all but the controller in order I'd like to figure this bit out. Thanks for your input, will check hector out...