ethz-asl/rotors_simulator #519: Not able to subscribe to camera msgs

shivamsardana, 2018-10-11:
OS : VMWare Ubuntu
ROS Kinetic

I am using bebop_autonomy package with RotorS package. I am able to subscribe to other rostopics but I am not able to subscribe camera topics. Evey time, let's say I try to view Bebop images in rviz, Gazebo get crashed. If I run rostopic info bebop2/camera_base/image_raw, it says nothing is published.

Can anybody here help me resolve this?

shivamsardana, 2018-10-11:
After subscribing to image_raw, gazebo crashes and also this messgae occurs to Gazebo terminal 

> "gzserver: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreRenderSystem.cpp:546: virtual void Ogre::RenderSystem::setDepthBufferFor(Ogre::RenderTarget*): Assertion `bAttached && A new DepthBuffer for a RenderTarget was created, but after creation" "it says it's incompatible with that RT"' failed.
Aborted (core dumped)

ffurrer, 2018-12-06:
I don't think this issue is specific to rotors, but rather an issue with ogre an gazebo, please post it there and state what system setup you are using. (OS, how you installed things, ...)

closed 2018-12-06 by ffurrer