ethz-asl/rotors_simulator #531: catkin_build error rotors_gazebo_plugins fails to build

wuyuzaizai, 2018-12-06:
[libprotobuf WARNING /home/zhaojiajia/cartographer/protobuf/src/google/protobuf/compiler/parser.cc:546] No syntax specified for the proto file: quaternion.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
In file included from /home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Wrench.pb.h:32:0,
                 from /home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Wrench.pb.cc:5:
/usr/include/gazebo-7/gazebo/msgs/vector3d.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/usr/include/gazebo-7/gazebo/msgs/vector3d.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/usr/include/gazebo-7/gazebo/msgs/vector3d.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^
In file included from /home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Imu.pb.h:32:0,
                 from /home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Imu.pb.cc:5:
/usr/include/gazebo-7/gazebo/msgs/quaternion.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/usr/include/gazebo-7/gazebo/msgs/quaternion.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/usr/include/gazebo-7/gazebo/msgs/quaternion.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^
In file included from /home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Imu.pb.h:33:0,
                 from /home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Imu.pb.cc:5:
/usr/include/gazebo-7/gazebo/msgs/vector3d.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/usr/include/gazebo-7/gazebo/msgs/vector3d.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/usr/include/gazebo-7/gazebo/msgs/vector3d.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^
In file included from /home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Wrench.pb.cc:5:0:
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Wrench.pb.h: In member function ‘const gazebo::msgs::Vector3d& gz_geometry_msgs::Wrench::force() const’:
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Wrench.pb.h:212:8: error: ‘_Vector3d_default_instance_’ is not a member of ‘gazebo::msgs’
       &::gazebo::msgs::_Vector3d_default_instance_);
        ^
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Wrench.pb.h: In member function ‘const gazebo::msgs::Vector3d& gz_geometry_msgs::Wrench::torque() const’:
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Wrench.pb.h:258:8: error: ‘_Vector3d_default_instance_’ is not a member of ‘gazebo::msgs’
       &::gazebo::msgs::_Vector3d_default_instance_);
        ^
In file included from /home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Imu.pb.cc:5:0:
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Imu.pb.h: In member function ‘const gazebo::msgs::Quaternion& gz_sensor_msgs::Imu::orientation() const’:
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Imu.pb.h:326:8: error: ‘_Quaternion_default_instance_’ is not a member of ‘gazebo::msgs’
       &::gazebo::msgs::_Quaternion_default_instance_);
        ^
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Imu.pb.h: In member function ‘const gazebo::msgs::Vector3d& gz_sensor_msgs::Imu::angular_velocity() const’:
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Imu.pb.h:402:8: error: ‘_Vector3d_default_instance_’ is not a member of ‘gazebo::msgs’
       &::gazebo::msgs::_Vector3d_default_instance_);
        ^
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Imu.pb.h: In member function ‘const gazebo::msgs::Vector3d& gz_sensor_msgs::Imu::linear_acceleration() const’:
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Imu.pb.h:478:8: error: ‘_Vector3d_default_instance_’ is not a member of ‘gazebo::msgs’
       &::gazebo::msgs::_Vector3d_default_instance_);
        ^
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Wrench.pb.cc: In static member function ‘static void gz_geometry_msgs::protobuf_Wrench_2eproto::TableStruct::InitDefaultsImpl()’:
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Wrench.pb.cc:95:19: error: ‘gazebo::msgs::protobuf_vector3d_2eproto’ has not been declared
   ::gazebo::msgs::protobuf_vector3d_2eproto::InitDefaults();
                   ^
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Wrench.pb.cc:99:7: error: ‘internal_default_instance’ is not a member of ‘gazebo::msgs::Vector3d’
       ::gazebo::msgs::Vector3d::internal_default_instance());
       ^
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Wrench.pb.cc:101:7: error: ‘internal_default_instance’ is not a member of ‘gazebo::msgs::Vector3d’
       ::gazebo::msgs::Vector3d::internal_default_instance());
       ^
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Wrench.pb.cc: In function ‘void gz_geometry_msgs::protobuf_Wrench_2eproto::{anonymous}::AddDescriptorsImpl()’:
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Wrench.pb.cc:121:19: error: ‘gazebo::msgs::protobuf_vector3d_2eproto’ has not been declared
   ::gazebo::msgs::protobuf_vector3d_2eproto::AddDescriptors();
                   ^
[... 20 lines cut ...]
/home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins/Imu.pb.cc:144:19: error: ‘gazebo::msgs::protobuf_vector3d_2eproto’ has not been declared
   ::gazebo::msgs::protobuf_vector3d_2eproto::AddDescriptors();
                   ^
make[2]: *** [CMakeFiles/mav_msgs.dir/Wrench.pb.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/mav_msgs.dir/Imu.pb.cc.o] Error 1
make[1]: *** [CMakeFiles/mav_msgs.dir/all] Error 2
make: *** [all] Error 2
cd /home/zhaojiajia/catkin_ws_simu/build/rotors_gazebo_plugins; catkin build --get-env rotors_gazebo_plugins | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << rotors_gazebo_plugins:make           [ Exited with code 2 ]      
Failed    <<< rotors_gazebo_plugins                [ 7.2 seconds ]             
Abandoned <<< mavros_extras                        [ Unrelated job failed ]    
Abandoned <<< rotors_gazebo                        [ Unrelated job failed ]    
Abandoned <<< rotors_hil_interface                 [ Unrelated job failed ]    
Abandoned <<< test_mavros                          [ Unrelated job failed ]    
Finished  <<< bebop_tools                          [ 3.2 seconds ]             
Finished  <<< mavros                               [ 1 minute and 2.9 seconds ]
[build] Summary: 21 of 26 packages succeeded.                                  
[build]   Ignored:   3 packages were skipped or are blacklisted.               
[build]   Warnings:  7 packages succeeded with warnings.                       
[build]   Abandoned: 4 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 1 minute and 53.6 seconds total.                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.

raymond30031, 2019-03-26:
what is your protobuf version?
type protoc --version

IhabMohamed, 2019-03-26:
Hi @raymond30031 

I have the same error and my protoc --version is "libprotoc 3.5.1"

Could you please help me?
Regards,
Ihab

ToniRV, 2019-03-27:
Same with "libprotoc 3.6.1"

raymond30031, 2019-03-27:
So Gazebo is built against the version of protoc released to the ubuntu repos and for 16.04 the protoc version is 2.6.1.

My setup is Ubuntu 16.04, Kinetic, and Gazebo 9. I am using 9 instead of 7 because i need it, but I think the default 7 should work after you make the protoc version switch. If you are using gazebo 9, don't forget to switch to the feature/gazebo9 branch.

You can follow the answer by MBD in the reference link. In the second step, I would not remove, but backup the /usr/local/include/google/protobuf and the /usr/local/bin/protoc in case you want to switch back.

Reference:
http://answers.gazebosim.org/question/18096/this-file-was-generated-by-an-older-version-of-protoc/

IhabMohamed, 2019-03-28:
Dear @raymond30031,

I would like to thank you for sharing this link. It was very useful and solved the problem.  I had to delete the installed version (i.e., libprotoc 3.5.1) and install the recommended one.  

Have a nice day!
Ihab

michaelpantic, 2019-04-10:
Closing as this seems to be solved - nice!

closed 2019-04-10 by michaelpantic

pkyadav73199, 2020-05-17:
Hi @raymond30031

I have the same error and my protoc --version is "libprotoc 2.6.1 "

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial

gazebo7      7.16.0-1~xenial

CMake Error at /home/edith/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins/cmake/FindGlog.cmake:77 (MESSAGE):
  Failed to find glog - Could not find glog include directory, set
  GLOG_INCLUDE_DIR to directory containing glog/logging.h
Call Stack (most recent call first):
  cmake/FindGlog.cmake:103 (GLOG_REPORT_NOT_FOUND)
  CMakeLists.txt:147 (find_package)

Could you please help me?

Regard
Pintu

surfii3z, 2020-09-03:
> Same with `libprotoc 3.6.1`
> I downgraded everything to `libprotoc 3.0.0` which is the official bionic version.
> 
> ```
> dpkg -l | grep protobuf
> ii  libprotobuf-dev:amd64                         3.0.0-9.1ubuntu1                             amd64        protocol buffers C++ library (development files)
> ii  libprotobuf-lite10:amd64                      3.0.0-9.1ubuntu1                             amd64        protocol buffers C++ library (lite version)
> ii  libprotobuf-lite17:amd64                      3.6.1-0lambda1                               amd64        protocol buffers runtime library (lite version)
> ii  libprotobuf10:amd64                           3.0.0-9.1ubuntu1                             amd64        protocol buffers C++ library
> ii  libprotobuf17:amd64                           3.6.1-0lambda1                               amd64        protocol buffers runtime library
> ii  protobuf-compiler                             3.0.0-9.1ubuntu1                             amd64        compiler for protocol buffer definition files
> ii  python-protobuf                               3.0.0-9.1ubuntu1                             amd64        Python bindings for protocol buffers
> ii  python3-protobuf                              3.0.0-9.1ubuntu1                             amd64        Python 3 bindings for protocol buffers
> ```
> 
> ```
> dpkg -l | grep libprotoc
> ii  libprotoc-dev:amd64                           3.0.0-9.1ubuntu1                             amd64        protocol buffers compiler library (development files)
> ii  libprotoc10:amd64                             3.0.0-9.1ubuntu1                             amd64        protocol buffers compiler library
> ii  libprotoc17:amd64                             3.6.1-0lambda1                               amd64        protocol buffers compiler runtime library
> ```
> 
> Now it seems to work, my system:
> 
> ```
> lsb_release -a
> No LSB modules are available.
> Distributor ID:	Ubuntu
> Description:	Ubuntu 18.04.2 LTS
> Release:	18.04
> Codename:	bionic
> ```
> 
> ```
> dpkg -l | grep gazebo
> ii  gazebo9                                       9.8.0-1~bionic                               amd64        Open Source Robotics Simulator
> ii  gazebo9-common                                9.8.0-1~bionic                               all          Open Source Robotics Simulator - Shared files
> ii  gazebo9-plugin-base                           9.8.0-1~bionic                               amd64        Open Source Robotics Simulator - base plug-ins
> ii  libgazebo9:amd64                              9.8.0-1~bionic                               amd64        Open Source Robotics Simulator - shared library
> ii  libgazebo9-dev:amd64                          9.8.0-1~bionic                               amd64        Open Source Robotics Simulator - Development Files
> ii  ros-melodic-gazebo-dev                        2.8.4-0bionic.20180706.234303                amd64        Provides a cmake config for the default version of Gazebo for the ROS distribution.
> ii  ros-melodic-gazebo-msgs                       2.8.4-0bionic.20181112.202939                amd64        Message and service data structures for interacting with Gazebo from ROS.
> ii  ros-melodic-gazebo-plugins                    2.8.4-0bionic.20181117.185301                amd64        Robot-independent Gazebo plugins for sensors, motors and dynamic reconfigurable components.
> ```

I can confirm that this works on Ubuntu 18.04 LTS with ROS Melodic.

To install protoc 3.0.0, one can follow this  [guide](https://github.com/linux-on-ibm-z/docs/wiki/Building-ProtoBuf-3.0.0)