ethz-asl/rotors_simulator #555: rosservice "/world/get_octomap" not getting published

viswanarayanans, 2019-03-13:
I am trying to make an octomap. I have used the following command for launching the gazebo simulation
`roslaunch rotors_gazebo mav_hovering_example.launch world_name:=waypoint`
I found out from [](https://github.com/ethz-asl/rotors_simulator/wiki/Generate-an-octomap-from-your-world) that octomap can be generated if the following line is present in the world file.
`<plugin name='gazebo_octomap' filename='librotors_gazebo_octomap_plugin.so'/>`
The file describes the octomap service and published topic as follows:
```
      <octomapPubTopic>world/octomap</octomapPubTopic>
      <octomapServiceName>world/get_octomap</octomapServiceName>
```
After launching, the simulation works fine, except that when I use the command `rosservice list` or `rostopic list`, I am unable to see the service and topic.

Please let me know, if I am missing anything.

juanmaca9, 2019-05-10:
I have the same problem. Did you find any solution for that? 
Thanks.

paul218, 2019-05-23:
You should check the cmakefile, and here there are options for compile the plugins, octomap plugin is  false , change to true and compile again