ethz-asl/rotors_simulator #556: ros melodic/gazebo9 package did not worked

aeudes, 2019-03-19:
On fresh installation on 18.04 with Gazebo9 and ros melodic, the rotors packages are non functional:
when I try to launch any example, the mav did not take off. After investigation, I have find that gazebo complain that:
```
librotors_gazebo_interface_plugin.so cound not find libmav_msgs.so
```

The package ros-melodic-rotors-gazebo-plugin did not contain such library but it should as it is defined here :
https://github.com/ethz-asl/rotors_simulator/blob/7e25747336be6e5413e819ecf5ebb176d9b55367/rotors_gazebo_plugins/CMakeLists.txt#L199-L212
Please note that this lib is not related to the package mav_msg but is the shared linked library that contained protobuf definition for internal rotors messages.

This previous PR #506 shows the same problem and should correct this issue.

tomcattigerkkk, 2020-02-20:
Can you post the caktin build error msg. I think you may have another compile problems

gsilano, 2020-02-20:
Hi all! As you can see from #506, I solved the problem (at least in my case). Please, take a look at my [CrazyS](https://github.com/gsilano/CrazyS) and [BebopS](https://github.com/gsilano/BebopS) repositories.

closed 2021-09-25 by Jaeyoung-Lim