ethz-asl/rotors_simulator #558: the format of image

DRAEYE, 2019-04-02:
Hello, after I mounted a camera on the UAV, the image it published were all gray level images, what I need to do to get the color image, is it associated with the parameter "image_format"?

showaykerker, 2019-06-10:
Here's how I deal with it:

Edit `rotors_simulator/rotors_description/urdf/component_snippets.xacro`
1. Change .so file of the plugin
```xml
<robot xmlns:xacro="http://ros.org/wiki/xacro">
	...
	<!-- Macro to add a camera. -->
	<xacro:macro name="camera_macro"
	    params="namespace parent_link camera_suffix frame_rate
	    horizontal_fov image_width image_height image_format min_distance
	    max_distance noise_mean noise_stddev enable_visual *geometry *origin">
	    	...
	    	<gazebo reference="${namespace}/camera_${camera_suffix}_link">
	    		<sensor type="camera" name="${namespace}_camera_${camera_suffix}">
	    			...
	    			<plugin name="${namespace}_camera_${camera_suffix}_controller" filename="librotors_gazebo_noisydepth_plugin.so">
	    				...
	    			</plugin>
                                 ...
	    		</sensor>
                    	...
		</gazebo>
		...
	</xacro:macro>
	...
</robot>

```
2. Change image format to R8G8B8, changing only of them should work fine.
```xml
	
<robot xmlns:xacro="http://ros.org/wiki/xacro">
	...
		
	<!-- Option 1 -->
	<!-- This affects topic ${namespace}/hummingbird/vi_sensor/left/image_raw -->

	<!-- Macro to add a VI-sensor stereo camera. -->
	<xacro:macro name="vi_sensor_stereo_camera_macro"
		   params="namespace parent_link frame_rate origin_offset_x baseline_y origin_offset_z max_range">
		<xacro:stereo_camera_macro
		    ...
		    image_format="R8G8B8"
		    .../>
		...
	</xacro:macro>

		...

	<!-- Option 2 -->
	<!-- This affects topic  ${namespace}/vi_sensor/camera_depth/camera/image_raw -->

	<!-- Macro to add a depth camera on the VI-sensor. -->
	<xacro:macro name="vi_sensor_depth_macro"
	    params="namespace parent_link camera_suffix frame_rate max_range *origin">
		...
		<gazebo reference="${namespace}/camera_${camera_suffix}_link">
	    		<sensor type="depth" name="${namespace}_camera_{camera_suffix}">
	    			...
	    			<camera>
	    				...
	    				<image>
	    					...
	    					<format>R8G8B8</format>>
	    					...
	    				</image>
	    				...
	    			</camera>
	    			...
	    		</sensor>
	    		...
	    	</gazebo>
	    	...
	</xacro:macro>
	...
</robot>
	
```

juanmed, 2019-06-14:
Adding to the answer by @showaykerker, when adding the xacro to your .gazebo mav model, one can simply change the ```image_format``` parameter to ```R8G8B8```:

```xml
  <xacro:camera_macro
    namespace="${namespace}"
    parent_link="${namespace}/base_link"
    camera_suffix="nadir"
    frame_rate="20.0"
    horizontal_fov="2"
    image_width="752"
    image_height="480"
    image_format="R8G8B8"
    min_distance="0.02"
    max_distance="50"
    noise_mean="0.0"
    noise_stddev="0.007"
    enable_visual="true">
    <box size="0.05 0.05 0.02" />
    <origin xyz="0 0 ${arm_length/2}" rpy="0 0 ${pi/4}" />
  </xacro:camera_macro>
```