ethz-asl/rotors_simulator #559: Mav Hovering Example not working on ROS Melodic in Ubuntu 18.04

z3r0f4z3, 2019-04-03:
After installing ROS melodic on Gazebo 9 and RotorS, im trying to run the mav_hovering example . It loads but the firefly is stuck and cannot move and Terminal gave the following log when running this ;
$ roslaunch rotors_gazebo mav_hovering_example.launch mav_name:=firefly world_name:=basic

The resulting log is :
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global property: pi
when processing file: /home/f4ze/catkin_ws/src/rotors_simulator/rotors_description/urdf/component_snippets.xacro
included from: /home/f4ze/catkin_ws/src/rotors_simulator/rotors_description/urdf/firefly_base.xacro
included from: /home/f4ze/catkin_ws/src/rotors_simulator/rotors_description/urdf/mav_generic_odometry_sensor.gazebo
xacro.py is deprecated; please use xacro instead
started roslaunch server http://Superion:35397/

SUMMARY
========

PARAMETERS
 * /firefly/lee_position_controller_node/angular_rate_gain/x: 0.52
 * /firefly/lee_position_controller_node/angular_rate_gain/y: 0.52
 * /firefly/lee_position_controller_node/angular_rate_gain/z: 0.18
 * /firefly/lee_position_controller_node/attitude_gain/x: 3
 * /firefly/lee_position_controller_node/attitude_gain/y: 3
 * /firefly/lee_position_controller_node/attitude_gain/z: 0.15
 * /firefly/lee_position_controller_node/inertia/xx: 0.0347563
 * /firefly/lee_position_controller_node/inertia/xy: 0.0
 * /firefly/lee_position_controller_node/inertia/xz: 0.0
 * /firefly/lee_position_controller_node/inertia/yy: 0.0458929
 * /firefly/lee_position_controller_node/inertia/yz: 0.0
 * /firefly/lee_position_controller_node/inertia/zz: 0.0977
 * /firefly/lee_position_controller_node/mass: 1.56779
 * /firefly/lee_position_controller_node/position_gain/x: 6
 * /firefly/lee_position_controller_node/position_gain/y: 6
 * /firefly/lee_position_controller_node/position_gain/z: 6
 * /firefly/lee_position_controller_node/rotor_configuration/0/angle: 0.52359877559
 * /firefly/lee_position_controller_node/rotor_configuration/0/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/0/direction: 1.0
 * /firefly/lee_position_controller_node/rotor_configuration/0/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/0/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/1/angle: 1.57079632679
 * /firefly/lee_position_controller_node/rotor_configuration/1/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/1/direction: -1.0
 * /firefly/lee_position_controller_node/rotor_configuration/1/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/1/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/2/angle: 2.61799387799
 * /firefly/lee_position_controller_node/rotor_configuration/2/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/2/direction: 1.0
 * /firefly/lee_position_controller_node/rotor_configuration/2/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/2/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/3/angle: -2.61799387799
 * /firefly/lee_position_controller_node/rotor_configuration/3/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/3/direction: -1.0
 * /firefly/lee_position_controller_node/rotor_configuration/3/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/3/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/4/angle: -1.57079632679
 * /firefly/lee_position_controller_node/rotor_configuration/4/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/4/direction: 1.0
 * /firefly/lee_position_controller_node/rotor_configuration/4/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/4/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/5/angle: -0.52359877559
 * /firefly/lee_position_controller_node/rotor_configuration/5/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/5/direction: -1.0
 * /firefly/lee_position_controller_node/rotor_configuration/5/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/5/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/velocity_gain/x: 4.7
 * /firefly/lee_position_controller_node/velocity_gain/y: 4.7
 * /firefly/lee_position_controller_node/velocity_gain/z: 4.7
 * /firefly/robot_description: <?xml version="1....
 * /firefly/tf_prefix: 
 * /gazebo/enable_ros_network: True
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /use_sim_time: True

NODES
  /firefly/
    hovering_example (rotors_gazebo/hovering_example)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    lee_position_controller_node (rotors_control/lee_position_controller_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_firefly (gazebo_ros/spawn_model)
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)

auto-starting new master
process[master]: started with pid [13699]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 41658e38-55e3-11e9-b104-1c872cb4886b
process[rosout-1]: started with pid [13710]
started core service [/rosout]
process[gazebo-2]: started with pid [13717]
process[gazebo_gui-3]: started with pid [13722]
process[firefly/spawn_firefly-4]: started with pid [13727]
ERROR: cannot launch node of type [rotors_control/lee_position_controller_node]: can't locate node [lee_position_controller_node] in package [rotors_control]
ERROR: cannot launch node of type [rotors_gazebo/hovering_example]: can't locate node [hovering_example] in package [rotors_gazebo]
process[firefly/robot_state_publisher-7]: started with pid [13728]
process[firefly/joint_state_publisher-8]: started with pid [13729]
[ INFO] [1554277017.266055300]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1554277017.266877732]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1554277017.306950555]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1554277017.308120411]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[INFO] [1554277017.750699, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1554277017.765395, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1554277018.369844, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1554277018.723012, 0.000000]: Spawn status: SpawnModel: Successfully spawned entity
[firefly/spawn_firefly-4] process has finished cleanly
log file: /home/f4ze/.ros/log/41658e38-55e3-11e9-b104-1c872cb4886b/firefly-spawn_firefly-4*.log
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'
[ INFO] [1554277025.702637804, 0.030000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1554277025.724685690, 0.050000000]: Physics dynamic reconfigure ready.

Alternatively does anyone know how to get RotorS to work on ROS Melodic on Gazebo 9 in Ubuntu 18.04 LTS. I am a first time user so I don't know much, I also don't if this specific issue has been raised already.

bys1123, 2019-04-03:
I have this error too, seems api.ignitionfuel.org is broken.

michaelpantic, 2019-04-10:
Hi! 

From your description I assume you installed rotorS from the Ros packages? These are still in a preliminary state. So I recommend you try it with the source version from this repo - thanks!

See https://github.com/ethz-asl/rotors_simulator/pull/506 for more details.

golden-memory, 2020-07-24:
I have this too, can someone help me