ethz-asl/rotors_simulator #565: Catkin build errors with mavlink/mavros (kinetic)

JeffR1992, 2019-05-22:
I followed the installation instructions for Ubuntu 16.04 with ROS Kinetic (https://github.com/ethz-asl/rotors_simulator), and have run into various errors when using `catkin build`, which seem to be related to the mavlink/mavros package. The initial build error that I got after attempting my first `catkin build` was the following:

[image]

This was solved by installing the `future` package. However, on trying to build again, I got the following error:

[image]

Which was solved by installing the required package using `apt-get install libgeographic-dev`. However, when I try and build again, I'm still getting the following error, which I've unfortunately been unable to solve:

[image]

This most recent error seems to be in the code, and having done some searching, I'm not sure how I can fix the `twist_covariance` error message. Any help would be greatly appreciated.

P.S. I had assumed that, since I installed a clean version of Ubuntu 16.04, and followed the RotorS installation instructions to a tee (twice, since I thought I did something wrong the first time) such errors would not have occurred.

EmbersArc, 2019-05-22:
I went through the exact same steps as you did. Ended up just renaming `msg.twist_covariance` to `msg.velocity_covariance` on line 234. As far as I can tell it builds and runs fine. Curious what the right way to fix this would be though. I'm on 18.04 with ROS Melodic.

TSC21, 2019-05-22:
Update ros-*-mavlink

StrSpi, 2019-08-24:
I used the latest ‘marvos’ and have successfully solved the same problem.

LanWu076, 2020-04-16:
The latest marvos will solve the problem I guess. I'm using Melodic 18.04 and my rosinstall file is like this:

- git: {local-name: mav_comm, uri: 'https://github.com/ethz-asl/mav_comm.git'}
- git: {local-name: mavlink, uri: 'https://github.com/mavlink/mavlink-gbp-release.git', version: debian/melodic/mavlink}
- git: {local-name: mavros, uri: 'https://github.com/ethz-asl/mavros.git', version: 0880a97575f8c6b705096aca17ca04c6c6dcfc02}
- git: {local-name: glog_catkin, uri: 'https://github.com/ethz-asl/glog_catkin.git'}
- git: {local-name: mav_control_rw, uri: 'https://github.com/ethz-asl/mav_control_rw.git'}
- git: {local-name: rotors_simulator, uri: 'https://github.com/ethz-asl/rotors_simulator.git'}
- git: {local-name: numpy_eigen, uri: 'https://github.com/ethz-asl/numpy_eigen.git'}
- git: {local-name: catkin_boost_python_buildtool, uri: 'https://github.com/ethz-asl/catkin_boost_python_buildtool.git'}

Gallifery, 2020-05-27:
> Update ros-*-mavlink

Could you elaborate how to "Update ros-*-mavlink"?

cjmdd, 2022-03-18:
> 

Can you tell me where msg.twist_covariance is stored?

cjmdd, 2022-03-18:
> renaming `msg.twist_covariance` to `msg.velocity_covariance` on line 234
Can you tell me where msg.twist_covariance is stored?