ethz-asl/rotors_simulator #573: Execute a nominal trajectory without control?

morsingher, 2019-06-11:
Hi everyone, 

I'm wondering if it's possible to use the simulator in order to directly execute a given nominal trajectory without any control. I mean, given the position and orientation of the quadrotor at each time instant, is there a way to execute such trajectory directly, without a control system? I guess it may be possible by computing the nominal actuators' velocities corresponding to the trajectory, but I would like to have advices from anyone who will spend time with this. The reason behind this just to visualize the nominal trajectory and compare it to the real one.

Thank you in advance,

Marco.

juanmed, 2019-06-17:
@morsingher 

Using differential flatness based control you can achieve it. Basically, given a position and yaw angle trajectory, the rest of states( linear velocities, angular velocities, orientation) and all inputs can be calculated (rotor speeds). This is your "nominal" trajectory. This are then used as feed-forward terms to a linear controller (PID, LQR) which compensates for error. This is your 'real' trajectory. For a ready implementation, check RPG lab's  [quadrotor controller](https://github.com/uzh-rpg/rpg_quadrotor_control) which is the implementation of the ideas in [1].

Take a look at:
[1]Faessler, M., Franchi, A., & Scaramuzza, D. (2017). Differential Flatness of Quadrotor Dynamics Subject to Rotor Drag for Accurate Tracking of High-Speed Trajectories. https://doi.org/10.1109/LRA.2017.2776353
[2] Tal, E., & Karaman, S. (2019). Accurate Tracking of Aggressive Quadrotor Trajectories Using Incremental Nonlinear Dynamic Inversion and Differential Flatness. Proceedings of the IEEE Conference on Decision and Control, 2018–Decem, 4282–4288. https://doi.org/10.1109/CDC.2018.8619621

gsilano, 2019-06-18:
Hi @juanmed!! Thanks a lot for linking the repository and papers too. They seem very interesting for my research activity. I will enjoy them as soon as I can.