ethz-asl/rotors_simulator #580: Can't build rotors_gazebo_plugins with octomap

jyp0802, 2019-08-05:
**I'm using Gazebo 7 and ROS Kinetic**

I'm trying to use the octomap function as described in
https://github.com/ethz-asl/rotors_simulator/wiki/Generate-an-octomap-from-your-world

So I set **BUILD_OCTOMAP_PLUGIN** to **TRUE**
(I also had to fix the autobackporting file "seds.txt" because there was a "1" missing in this line
```s/(` |^|<|\()octoignition::math::Vector3d/\1octomath::Vector3/g```)

ANYWAYS, when I run catkin build, along with a bunch of warnings about things being deprecated, I get this error.
```
Errors     << rotors_gazebo_plugins:make /home/jyp0802/SLAM/gazebo_ws/logs/rotors_gazebo_plugins/build.make.003.log                                                                                                                                               
/home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_octomap_plugin.cpp: In member function ‘bool gazebo::OctomapFromGazeboWorld::ServiceCallback(rotors_comm::Octomap::Request&, rotors_comm::Octomap::Response&)’:
/home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_octomap_plugin.cpp:91:42: error: no matching function for call to ‘gazebo::common::SphericalCoordinates::SphericalFromLocal(gazebo::math::Vector3&)’
       SphericalFromLocal(origin_cartesian);
                                          ^
In file included from /usr/include/gazebo-7/gazebo/common/common.hh:40:0,
                 from /usr/include/gazebo-7/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
                 from /home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/include/rotors_gazebo_plugins/common.h:26,
                 from /home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/include/rotors_gazebo_plugins/gazebo_octomap_plugin.h:28,
                 from /home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_octomap_plugin.cpp:21:
/usr/include/gazebo-7/gazebo/common/SphericalCoordinates.hh:96:40: note: candidate: ignition::math::Vector3d gazebo::common::SphericalCoordinates::SphericalFromLocal(const Vector3d&) const
       public: ignition::math::Vector3d SphericalFromLocal(
                                        ^
/usr/include/gazebo-7/gazebo/common/SphericalCoordinates.hh:96:40: note:   no known conversion for argument 1 from ‘gazebo::math::Vector3’ to ‘const Vector3d& {aka const ignition::math::Vector3<double>&}’
make[2]: *** [CMakeFiles/rotors_gazebo_octomap_plugin.dir/src/gazebo_octomap_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/rotors_gazebo_octomap_plugin.dir/all] Error 2
make: *** [all] Error 2
```

This is the full output in the terminal
```
Errors     << rotors_gazebo_plugins:make /home/jyp0802/SLAM/gazebo_ws/logs/rotors_gazebo_plugins/build.make.000.log                                                                                                                                               
[libprotobuf WARNING google/protobuf/descriptor.cc:5411] Warning: Unused import: "MagneticField.proto" imports "quaternion.proto" which is not used.
/home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_gps_plugin.cpp: In member function ‘virtual void gazebo::GazeboGpsPlugin::Load(gazebo::sensors::SensorPtr, sdf::ElementPtr)’:
/home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_gps_plugin.cpp:45:60: warning: ‘std::__cxx11::string gazebo::sensors::Sensor::GetWorldName() const’ is deprecated [-Wdeprecated-declarations]
   world_ = physics::get_world(parent_sensor_->GetWorldName());
                                                            ^
In file included from /usr/include/gazebo-7/gazebo/sensors/AltimeterSensor.hh:25:0,
                 from /usr/include/gazebo-7/gazebo/sensors/sensors.hh:2,
                 from /home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/include/rotors_gazebo_plugins/gazebo_gps_plugin.h:27,
                 from /home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_gps_plugin.cpp:19:
/usr/include/gazebo-7/gazebo/sensors/Sensor.hh:200:27: note: declared here
       public: std::string GetWorldName() const GAZEBO_DEPRECATED(7.0);
                           ^
/home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_gps_plugin.cpp: In member function ‘void gazebo::GazeboGpsPlugin::OnUpdate()’:
/home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_gps_plugin.cpp:177:57: warning: ‘gazebo::common::Time gazebo::sensors::Sensor::GetLastMeasurementTime()’ is deprecated [-Wdeprecated-declarations]
   current_time = parent_sensor_->GetLastMeasurementTime();
                                                         ^
In file included from /usr/include/gazebo-7/gazebo/sensors/AltimeterSensor.hh:25:0,
                 from /usr/include/gazebo-7/gazebo/sensors/sensors.hh:2,
                 from /home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/include/rotors_gazebo_plugins/gazebo_gps_plugin.h:27,
                 from /home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_gps_plugin.cpp:19:
/usr/include/gazebo-7/gazebo/sensors/Sensor.hh:166:28: note: declared here
       public: common::Time GetLastMeasurementTime() GAZEBO_DEPRECATED(7.0);
                            ^
/home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_lidar_plugin.cpp: In member function ‘virtual void gazebo::GazeboLidarPlugin::Load(gazebo::sensors::SensorPtr, sdf::ElementPtr)’:
/home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_lidar_plugin.cpp:75:69: warning: ‘std::__cxx11::string gazebo::sensors::Sensor::GetWorldName() const’ is deprecated [-Wdeprecated-declarations]
   this->world = physics::get_world(this->parentSensor->GetWorldName());
                                                                     ^
In file included from /usr/include/gazebo-7/gazebo/sensors/RaySensor.hh:26:0,
                 from /home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/include/rotors_gazebo_plugins/external/gazebo_lidar_plugin.h:27,
                 from /home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_lidar_plugin.cpp:23:
/usr/include/gazebo-7/gazebo/sensors/Sensor.hh:200:27: note: declared here
       public: std::string GetWorldName() const GAZEBO_DEPRECATED(7.0);
                           ^
/home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_lidar_plugin.cpp:78:39: warning: ‘gazebo::physics::MultiRayShapePtr gazebo::sensors::RaySensor::GetLaserShape() const’ is deprecated [-Wdeprecated-declarations]
     this->parentSensor->GetLaserShape()->ConnectNewLaserScans(
                                       ^
[... 153 lines cut ...]
                 from /usr/include/gazebo-7/gazebo/plugins/CameraPlugin.hh:23,
                 from /opt/ros/kinetic/include/gazebo_plugins/gazebo_ros_camera_utils.h:50,
                 from /home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/include/rotors_gazebo_plugins/gazebo_noisydepth_plugin.h:35,
                 from /home/jyp0802/SLAM/gazebo_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_noisydepth_plugin.cpp:24:
/usr/include/gazebo-7/gazebo/sensors/Sensor.hh:166:28: note: declared here
       public: common::Time GetLastMeasurementTime() GAZEBO_DEPRECATED(7.0);
                            ^
make: *** [all] Error 2
cd /home/jyp0802/SLAM/gazebo_ws/build/rotors_gazebo_plugins; catkin build --get-env rotors_gazebo_plugins | catkin env -si  /usr/bin/make --jobserver-fds=14,18 -j; cd -
..................................................................................................................................................................................................................................................................
Failed     << rotors_gazebo_plugins:make           [ Exited with code 2 ]                                                                                                                                                                                         
Failed    <<< rotors_gazebo_plugins                [ 1 minute and 7.2 seconds ]                                                                                                                                           [... cut ...]
[build]   Failed:    1 packages failed.                                                                                                                                                                                                                           
[build] Runtime: 2 minutes and 15.4 seconds total.                                                                                                                                                                                                                
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

**Any ideas on how I can solve this?**

Thanks

YinshuaiSun, 2019-10-11:
hi! have you solved this problem, i met the same problem.

shiniankanchaige, 2019-12-16:
i faced the same.how  to solve it?

Wayne-xixi, 2020-03-09:
Hello, have you solved this problem, I met the same problem.