ethz-asl/rotors_simulator #584: catkin build error - rotors_gazebo_plugins failed

YinshuaiSun, 2019-10-11:
Errors     << rotors_gazebo_plugins:make /home/ariesys/RotorSim/logs/rotors_gazebo_plugins/build.make.003.log
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_gps_plugin.cpp: In member function ‘virtual void gazebo::GazeboGpsPlugin::Load(gazebo::sensors::SensorPtr, sdf::ElementPtr)’:
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_gps_plugin.cpp:45:47: error: ‘class gazebo::sensors::GpsSensor’ has no member named ‘GetWorldName’
   world_ = physics::get_world(parent_sensor_->GetWorldName());

/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_gps_plugin.cpp:177:34: error: ‘class gazebo::sensors::GpsSensor’ has no member named ‘GetLastMeasurementTime’
   current_time = parent_sensor_->GetLastMeasurementTime();
                                  ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_lidar_plugin.cpp: In member function ‘virtual void gazebo::GazeboLidarPlugin::Load(gazebo::sensors::SensorPtr, sdf::ElementPtr)’:
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_lidar_plugin.cpp:75:56: error: ‘class gazebo::sensors::RaySensor’ has no member named ‘GetWorldName’
   this->world = physics::get_world(this->parentSensor->GetWorldName());
                                                        ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_lidar_plugin.cpp:78:25: error: ‘class gazebo::sensors::RaySensor’ has no member named ‘GetLaserShape’
     this->parentSensor->GetLaserShape()->ConnectNewLaserScans(
                         ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_lidar_plugin.cpp:89:38: error: ‘class gazebo::sensors::Sensor’ has no member named ‘GetParentName’
   const string scopedName = _parent->GetParentName();
                                      ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_lidar_plugin.cpp: In member function ‘virtual void gazebo::GazeboLidarPlugin::OnNewLaserScans()’:
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_lidar_plugin.cpp:101:48: error: ‘class gazebo::sensors::RaySensor’ has no member named ‘GetRangeMin’
   lidar_message.set_min_distance(parentSensor->GetRangeMin());
                                                ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_lidar_plugin.cpp:102:48: error: ‘class gazebo::sensors::RaySensor’ has no member named ‘GetRangeMax’
   lidar_message.set_max_distance(parentSensor->GetRangeMax());
                                                ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_lidar_plugin.cpp:103:52: error: ‘class gazebo::sensors::RaySensor’ has no member named ‘GetRange’
   lidar_message.set_current_distance(parentSensor->GetRange(0));
                                                    ^
make[2]: *** [CMakeFiles/rotors_gazebo_gps_plugin.dir/src/gazebo_gps_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/rotors_gazebo_gps_plugin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_bag_plugin.cpp:375:58: error: ‘ignition::math::Vector3d {aka class ignition::math::Vector3<double>}’ has no member named ‘GetLength’
     double body1_force = contacts[i]->wrench->body1Force.GetLength();
                                                          ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_bag_plugin.cpp:387:65: error: ‘ignition::math::Vector3d {aka class ignition::math::Vector3<double>}’ has no member named ‘x’
     wrench_msg.wrench.force.x = contacts[i]->wrench->body1Force.x;
                                                                 ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_bag_plugin.cpp:388:65: error: ‘ignition::math::Vector3d {aka class ignition::math::Vector3<double>}’ has no member named ‘y’
     wrench_msg.wrench.force.y = contacts[i]->wrench->body1Force.y;
                                                                 ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_bag_plugin.cpp:389:65: error: ‘ignition::math::Vector3d {aka class ignition::math::Vector3<double>}’ has no member named ‘z’
     wrench_msg.wrench.force.z = contacts[i]->wrench->body1Force.z;
                                                                 ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_bag_plugin.cpp:390:67: error: ‘ignition::math::Vector3d {aka class ignition::math::Vector3<double>}’ has no member named ‘x’
     wrench_msg.wrench.torque.x = contacts[i]->wrench->body1Torque.x;
                                                                   ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_bag_plugin.cpp:391:67: error: ‘ignition::math::Vector3d {aka class ignition::math::Vector3<double>}’ has no member named ‘y’
     wrench_msg.wrench.torque.y = contacts[i]->wrench->body1Torque.y;
                                                                   ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_bag_plugin.cpp:392:67: error: ‘ignition::math::Vector3d {aka class ignition::math::Vector3<double>}’ has no member named ‘z’
     wrench_msg.wrench.torque.z = contacts[i]->wrench->body1Torque.z;
                                                                   ^
make[2]: *** [CMakeFiles/rotors_gazebo_lidar_plugin.dir/src/external/gazebo_lidar_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/rotors_gazebo_lidar_plugin.dir/all] Error 2
make[2]: *** [CMakeFiles/rotors_gazebo_bag_plugin.dir/src/gazebo_bag_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/rotors_gazebo_bag_plugin.dir/all] Error 2
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_geotagged_images_plugin.cpp: In member function ‘virtual void gazebo::GeotaggedImagesPlugin::Load(gazebo::sensors::SensorPtr, sdf::ElementPtr)’:
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_geotagged_images_plugin.cpp:64:40: error: ‘class gazebo::sensors::CameraSensor’ has no member named ‘GetCamera’
   this->camera_ = this->parentSensor_->GetCamera();
                                        ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_geotagged_images_plugin.cpp:72:28: error: ‘class gazebo::rendering::Scene’ has no member named ‘GetSimTime’
   lastImageTime_ = scene_->GetSimTime();
                            ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_geotagged_images_plugin.cpp:74:33: error: ‘class gazebo::rendering::Camera’ has no member named ‘GetImageWidth’
   this->width_ = this->camera_->GetImageWidth();
                                 ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_geotagged_images_plugin.cpp:75:34: error: ‘class gazebo::rendering::Camera’ has no member named ‘GetImageHeight’
   this->height_ = this->camera_->GetImageHeight();
                                  ^
[... 4 lines cut ...]
   this->format_ = this->camera_->GetImageFormat();
                                  ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_geotagged_images_plugin.cpp: In member function ‘void gazebo::GeotaggedImagesPlugin::OnNewFrame(const unsigned char*)’:
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_geotagged_images_plugin.cpp:133:26: error: ‘class gazebo::rendering::Camera’ has no member named ‘GetImageData’
   image = this->camera_->GetImageData(0);
                          ^
/home/ariesys/RotorSim/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_geotagged_images_plugin.cpp:135:38: error: ‘class gazebo::rendering::Scene’ has no member named ‘GetSimTime’
   common::Time currentTime = scene_->GetSimTime();
                                      ^
make[2]: *** [CMakeFiles/gazebo_geotagged_images_plugin.dir/src/external/gazebo_geotagged_images_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/gazebo_geotagged_images_plugin.dir/all] Error 2
make: *** [all] Error 2
[... cut ...]
[build]   Warnings:  None.                                                     
[build]   Abandoned: 1 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 3 minutes and 44.3 seconds total.   

It seems all the sensor plugins could not find theirs functions, is there something wrong with my gazebo. my gazebo version is gazebo 8, running on ubuntu 16.04 and ROS kinetic in VWware. Thanks very much!

shiniankanchaige, 2019-12-16:
have you sloved this problem？

ricardoGrando, 2020-02-20:
git reset --hard ac77a8a

That will do it.

ac77a8a ia a commit before "Remove older gazebo version functions." commit was performed.

491734045, 2020-05-04:
I have the same issue. Have you solved this problem?

closed 2021-01-19 by Jaeyoung-Lim