ethz-asl/rotors_simulator #586: gazebo_mavlink_interface not compiling

abono3, 2019-11-24:
[image]

Hello everyone! 
I am quite new to ROS and git so I apologize if there is already a solution for this problem that i couldn't find.
I am using Ubuntu 16.04 and gazebo 7. It seems that there is some package related to gazebo msgs missing. Can you help me?  I read about problems comping the hil plugin in ubuntu 16.04 but nothing about the gazebo_mavlink_interface.

Thank you for any help :)

antbono, 2019-11-29:
All the ros-kinetic-gazebo-* packages are installed on my system.

evanyeh5118, 2019-12-03:
I have the same problem. Is it related to the version of gazebo?

antbono, 2019-12-04:
No I don't think so. I tried with gazebo9 but it seems that the package is compatible only with gazzebo7. The 'fake' solution I adopted is installing the BebopS package (https://github.com/gsilano/BebopS) that is an extension of RotorS. Inside you can find all the RotorS files so you can use them once compiled.

evanyeh5118, 2019-12-05:
I see. So far my solution is to comment all the line with "gazebo::msgs::Any", then it could be built successfully. This is not the best choice but the basic usage like hovering launch file works well.
Thanks for your reply @antbono :)

shilongshen, 2019-12-07:
@evanyeh5118 ,你好，我也遇到了同样的问题，请问能告知我详细的解决方法吗？谢谢。
Hello, I have encountered the same problem. Could you tell me the detailed solution? Thank you.

gsilano, 2019-12-12:
Hello everyone! I followed the discussion and got the same problem when compiling RotorS with Gazebo 9. By downloading the latest version of the MAV_MSGS package the problem seems to disappear: I was able to run the example file (the launch file reported in the README.md file) either using the code in the master or the one in the branch features/gazebo9.

Since two of my repository ([BebopS](https://github.com/gsilano/BebopS) and [CrazyS)](https://github.com/gsilano/CrazyS) using RotorS as a basis, I will work to adapt them.

antbono, 2019-12-12:
Hello @gsilano! Thank you very much for this help! I tried following your advice but it didn't work for me because to problem was with the MAVROS package. I followed the RotorS [installation instructions for ubuntu16](https://github.com/ethz-asl/rotors_simulator/blob/master/README.md) then I substituted the MAVROS package downloaded with with the [latest one](https://github.com/mavlink/mavros). Now when compiling, I only have some warnings but none related to the MAV_MSGS package.

I really hope seeing BebopS and CrazyS updated with this compatibility with gazebo9. Do you think that this simple substitution will make them work? 

Thank you again!

gsilano, 2019-12-28:
Hi @antbono @shilongshen and @evanyeh5118!! As for the BebopS and CrazyS repositories, the problem that you arise, the compatibility of RotorS (and therefore BebopS) with ROS Melodic and Kinetic with Gazebo 9, I have just uploaded the instructions in the README.md of the packages. Do not hesitate to try my guide and highlight typing errors and / or errors.