ethz-asl/rotors_simulator #587: catkin build error following tutorial

AleBurzio, 2019-11-25:
Hi, I tried following the tutorial to install and use the rotorS package but catkin build keeps throwing errors
`
---------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic
Workspace:                   /home/alessandro/catkin_ws
---------------------------------------------------------------
Build Space:        [exists] /home/alessandro/catkin_ws/build
Devel Space:        [exists] /home/alessandro/catkin_ws/devel
Install Space:      [unused] /home/alessandro/catkin_ws/install
Log Space:         [missing] /home/alessandro/catkin_ws/logs
Source Space:       [exists] /home/alessandro/catkin_ws/src
DESTDIR:            [unused] None
---------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
---------------------------------------------------------------
[build] Found '21' packages in 0.0 seconds.                                    
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                 [ 2.0 seconds ]            
Starting  >>> mav_msgs                                                         
Starting  >>> mav_state_machine_msgs                                           
Starting  >>> mav_system_msgs                                                  
Starting  >>> mavlink                                                          
_______________________________________________________________________________
Errors     << mavlink:make /home/alessandro/catkin_ws/logs/mavlink/build.make.000.log
Traceback (most recent call last):
  File "/home/alessandro/catkin_ws/src/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/alessandro/catkin_ws/src/mavlink/pymavlink/generator/mavgen.py", line 12, in <module>
    from future import standard_library
ImportError: No module named future
make[2]: *** [uAvionix-v2.0-cxx-stamp] Error 1
make[1]: *** [CMakeFiles/uAvionix.xml-v2.0.dir/all] Error 2
make: *** [all] Error 2
cd /home/alessandro/catkin_ws/build/mavlink; catkin build --get-env mavlink | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << mavlink:make                          [ Exited with code 2 ]     
Failed    <<< mavlink                               [ 0.3 seconds ]            
Abandoned <<< mavros_msgs                           [ Unrelated job failed ]   
Abandoned <<< rotors_comm                           [ Unrelated job failed ]   
Abandoned <<< rotors_description                    [ Unrelated job failed ]   
Abandoned <<< rotors_evaluation                     [ Unrelated job failed ]   
Abandoned <<< mav_planning_msgs                     [ Unrelated job failed ]   
Abandoned <<< libmavconn                            [ Unrelated job failed ]   
Abandoned <<< mavros                                [ Unrelated job failed ]   
Abandoned <<< mavros_extras                         [ Unrelated job failed ]   
Abandoned <<< rotors_control                        [ Unrelated job failed ]   
Abandoned <<< rotors_gazebo_plugins                 [ Unrelated job failed ]   
Abandoned <<< rotors_gazebo                         [ Unrelated job failed ]   
Abandoned <<< rotors_hil_interface                  [ Unrelated job failed ]   
Abandoned <<< rotors_joy_interface                  [ Unrelated job failed ]   
Abandoned <<< rqt_rotors                            [ Unrelated job failed ]   
Abandoned <<< test_mavros                           [ Unrelated job failed ]   
Finished  <<< mav_system_msgs                       [ 6.1 seconds ]            
Finished  <<< mav_state_machine_msgs                [ 7.2 seconds ]            
Finished  <<< mav_msgs                              [ 8.1 seconds ]            
[build] Summary: 4 of 20 packages succeeded.                                   
[build]   Ignored:   2 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 15 packages were abandoned.                               
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 10.3 seconds total.                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

`

all the packages marked as abandoned are contained in the rotors_simulator folder within my catkin workspace, i am pretty new to this and I am unsure what i need to tweak to make it work
Thank you in advance, let me know if there is any other information I need to add

I am using ROS-melodic for ubuntu LTS 18.04

AleBurzio, 2019-11-25:
turns out i was missing modules, case solved

closed 2019-11-25 by AleBurzio

chanforg, 2020-03-01:
i also meet this error,how did you solve it?

JetJie, 2020-04-23:
I have same problem, can you tell me which modules you missed? thank you

AleBurzio11, 2020-04-23:
> I have same problem, can you tell me which modules you missed? thank you

I don't remeber exactly, it's been a few months, but for sure the "future" module
Try to install with bot pip and pip3  if you have multiple versions of python on your system, this applies to any other package you may have needed to install

gsilano, 2020-04-23:
Try with this

```console
$ cd catkin_ws
$ rosdep install --from-paths src -i
```

JetJie, 2020-04-23:
> > I have the same problem, can you tell me which modules you missed? thank you
> 
> I don't remeber exactly, it's been a few months, but for sure the "future" module
> Try to install with bot pip and pip3 if you have multiple versions of python on your system, this applies to any other package you may have needed to install

yes. I install the future by pip. It works, thank you. It strange that why this project only misses the module future.

JetJie, 2020-04-23:
> Try with this
> 
> ```
> $ cd catkin_ws
> $ rosdep install --from-paths src -i
> ```

I solved it by install the module future. thank you.