ethz-asl/rotors_simulator #590: rotors_gazebo_plugins

shilongshen, 2019-12-06:
when  i run catkin build ,something error happened.
[image]

anushmanukyan, 2019-12-20:
@shilongshen I was facing the same issue, but after following the steps described here: [https://github.com/gsilano/BebopS](url) solve the problem. Bebop is a new simulation based on RotorS, so rotors works perfectly.

gsilano, 2019-12-28:
Hi @shilongshen and @anushmanukyan!! As for the BebopS repository, the problem you arise, the compatibility of RotorS (and then BebopS) with ROS Melodic and Kinetic with Gazebo 9 (#586), I have just uploaded the instructions in the README.md of the package (BebopS). Don't hesitate to try my guide and highlight typos and / or mistakes.

LLlkaiwen, 2020-01-07:
I am using ROS Melodic and  solve this problem by "sudo apt-get install ros-melodic-mavlink"

Sweetyshadow, 2020-03-19:
> I am using ROS Melodic and solve this problem by "sudo apt-get install ros-melodic-mavlink"

it also works with Kinetic