ethz-asl/rotors_simulator #592: Erroro: rotors_gazebo_plugins build

alex-zador, 2019-12-14:
I have ubuntu 16.04, ros kinetic & gazebo 7.0 with Virtual Box.

Installed rotors_simulator following ReadMe.md
On 'catkin build' I've got:
#Errors     << rotors_gazebo_plugins:make /home/dron/catkin_ws/logs/rotors_gazebo_plugins/build.make.001.log      
#/home/dron/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_mavlink_interface.cpp: In member function ‘virtual void gazebo::GazeboMavlinkInterface::Load(gazebo::physics::ModelPtr, sdf::ElementPtr)’:
#/home/dron/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_mavlink_interface.cpp:124:41: error: ‘Any’ is not a member of ‘gazebo::msgs’
#                 node_handle_->Advertise<gazebo::msgs::Any>(gztopic_[index]);
#                                         ^
#/home/dron/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_mavlink_interface.cpp:124:41: error: ‘Any’ is not a member of ‘gazebo::msgs’
#/home/dron/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_mavlink_interface.cpp:124:75: error: no matching function for call to ‘gazebo::transport::Node::Advertise(std::__cxx11::string&)’
#                 node_handle_->Advertise<gazebo::msgs::Any>(gztopic_[index]);
#                                                                           ^
#In file included from /usr/include/gazebo-7/gazebo/transport/transport.hh:6:0,
#                 from /usr/include/gazebo-7/gazebo/gazebo_core.hh:22,
#                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
#                 from /home/dron/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins/include/rotors_gazebo_plugins/gazebo_mavlink_interface.h:49,
#                 from /home/dron/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_mavlink_interface.cpp:22:
#...

#Failed     << rotors_gazebo_plugins:make            [ Exited with code 2 ]                                       
#Failed    <<< rotors_gazebo_plugins                 [ 31.4 seconds ]                                             
#Abandoned <<< rotors_hil_interface                  [ Unrelated job failed ]                                     
#Abandoned <<< rotors_gazebo                         [ Unrelated job failed ]                                     
#Abandoned <<< test_mavros                           [ Unrelated job failed ]                                     
#Finished  <<< mavros_extras                         [ 1 minute and 57.2 seconds ]                                
#[build] Summary: 15 of 19 packages succeeded.                                                                    
#[build]   Ignored:   2 packages were skipped or are blacklisted.                                                 
#[build]   Warnings:  None.                                                                                       
#[build]   Abandoned: 3 packages were abandoned.                                                                  
#[build]   Failed:    1 packages failed. 

Please, help! Will upgrade to ubuntu 18, ros melodic (gazebo 9) help?

alex-zador, 2019-12-15:
Problem has been solved. 
File rotors_simulator/rotors_gazebo_plugins/src/gazebo_mavlink_interface.cpp rolled back to restore old gazebo support. catkin build ran without errors.

[build] Summary: All 20 packages succeeded!                                                                                                           
[build]   Ignored:   2 packages were skipped or are blacklisted.                                                                                      
[build]   Warnings:  4 packages succeeded with warnings.                                                                                              
[build]   Abandoned: None.                                                                                                                            
[build]   Failed:    None.                                                                                                                            
[build] Runtime: 25 minutes and 10.5 seconds total.                                                                                                   
[build] Note: Workspace packages have changed, please re-source setup files to use them.

closed 2019-12-15 by alex-zador

shiniankanchaige, 2019-12-16:
> Problem has been solved.
> File rotors_simulator/rotors_gazebo_plugins/src/gazebo_mavlink_interface.cpp rolled back to restore old gazebo support. catkin build ran without errors.
>
> [build] Summary: [...]

i face the same problem 。how to solve it？

fthgkc, 2019-12-30:
Inside rotors_simulator folder 

git reset --hard ac77a8a

command solved my problem. 

ac77a8a is coming from https://github.com/ethz-asl/rotors_simulator/commits/master corresponding to an earlier commit before "Remove older gazebo version functions." commit.

gsilano, 2019-12-30:
In #590 my solution.

jungx148, 2020-02-17:
@gsilano The above problem is solved and "catkin build" compiles well. Thank you very much.

https://github.com/gsilano/BebopS

CoolLoveBoy, 2020-04-03:
> File rotors_simulator/rotors_gazebo_plugins/src/gazebo_mavlink_interface.cpp rolled back to restore old gazebo support. catkin build ran without errors.

I have the same problem. How to solve it?

w407022008, 2020-07-17:
@gsilano Would you mind give me some suggestion, in [#632](https://github.com/ethz-asl/rotors_simulator/issues/632)? thx in advance.

gsilano, 2020-07-17:
Have you followed all the steps reported in my procedure? Have you updated Gazebo?

w407022008, 2020-07-17:
@gsilano Thx Giuseppe! Actually i did follow all the steps, and the Gazebo has been updated up to version-9. Well, just right now, i deleted all and re-installed again it in a fully new catkin_ws. the error appears that 

> fatal error: google/protobuf/port_def.inc: No such file or directory
 #include <google/protobuf/port_def.inc>

and some warning:

> CMake Warning at /usr/share/cmake-3.10/Modules/FindProtobuf.cmake:455 (message):
  Protobuf compiler version 3.7.1 doesn't match library version 3.4.0
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:165 (find_package)
  CMakeLists.txt:113 (find_package)

CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindBoost.cmake:911 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "chrono" will no longer be dereferenced when the
  policy is set to NEW.  Since the policy is not set the OLD behavior will be
  used.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:1558 (_Boost_MISSING_DEPENDENCIES)
  /usr/share/OGRE/cmake/modules/FindOGRE.cmake:318 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:175 (find_package)
  CMakeLists.txt:113 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning at /usr/share/cmake-3.10/Modules/FindProtobuf.cmake:455 (message):
  Protobuf compiler version 3.7.1 doesn't match library version 3.4.0
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/ignition-cmake0/cmake0/FindIgnProtobuf.cmake:31 (find_package)
  /usr/share/cmake-3.10/Modules/CMakeFindDependencyMacro.cmake:48 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/ignition-transport4/ignition-transport4-config.cmake:106 (find_dependency)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:197 (find_package)
  CMakeLists.txt:113 (find_package)

CMake Warning at /usr/share/cmake-3.10/Modules/FindProtobuf.cmake:455 (message):
  Protobuf compiler version 3.7.1 doesn't match library version 3.4.0
Call Stack (most recent call first):
  CMakeLists.txt:179 (find_package)

CMake Warning at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'gazebo' but neither 'gazebo_INCLUDE_DIRS' nor
  'gazebo_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:231 (catkin_package)

CMake Warning at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'octomap' but neither 'octomap_INCLUDE_DIRS'
  nor 'octomap_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:231 (catkin_package)

CMake Warning at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'opencv' but neither 'opencv_INCLUDE_DIRS' nor
  'opencv_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:231 (catkin_package)

cd /home/sique/catkin_ws/build/rotors_gazebo_plugins; catkin build --get-env rotors_gazebo_plugins | catkin env -si  /usr/bin/make cmake_check_build_system; cd -

When i `sudo apt-get remove protobuf-compiler` and `sudo apt-get install protobuf-compiler` again, another error appears that 

> CMake Error at /home/sique/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins/CMakeLists.txt:113 (find_package):
  By not providing "Findgazebo.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "gazebo", but
  CMake did not find one.

  Could not find a package configuration file provided by "gazebo" with any
  of the following names:

    gazeboConfig.cmake
    gazebo-config.cmake

  Add the installation prefix of "gazebo" to CMAKE_PREFIX_PATH or set
  "gazebo_DIR" to a directory containing one of the above files.  If "gazebo"
  provides a separate development package or SDK, be sure it has been
  installed.

make: *** [cmake_check_build_system] Error 1
cd /home/sique/catkin_ws/build/rotors_gazebo_plugins; catkin build --get-env rotors_gazebo_plugins | catkin env -si  /usr/bin/make cmake_check_build_system; cd -

Sequenly, after i `sudo apt-get install -y libgazebo9-dev`, the first error appears again.

gsilano, 2020-07-17:
Hmmm. I suggest you take a look at the issues already open on BebopS. This problem seems to me already discussed.

damianosa96, 2020-12-31:
Hello, 
I'm having a problem with catkin build, in particular with `rotors_gazebo_plugins:make`. 
I have Ubuntu 16.04 and Gazebo 7, that are the recommended versions to install RotorS.

I tried to follow the @fthgkc suggestion (git reset --hard ac77a8a), but it didn't work, the issue remains. 

Here is the error I receive:

[image]
[image]

It seems there is something missing in some libraries, but I can't guess what it is.

Any suggestions?

Thanks

gsilano, 2020-12-31:
Hi @damianosa96. Please, see this issue #590  and the message reported above.

> In #590 my solution.

Sunshine979797, 2024-06-05:
> Problem has been solved. File rotors_simulator/rotors_gazebo_plugins/src/gazebo_mavlink_interface.cpp rolled back to restore old gazebo support. catkin build ran without errors.
>
> [build] Summary: [...]