ethz-asl/rotors_simulator #594: Building on Ubuntu 16.04 with ROS kinetic Fails

MohammadKassemZein, 2020-01-27:
Hello,
I followed the instructions carefully, but the package does not seem to work with Ubuntu 16.04.
These are the errors I get:

Errors     << mavros_extras:make /home/zein/catkin_pelican/logs/mavros_extras/build.make.000.log
/home/zein/catkin_pelican/src/mavros/mavros_extras/src/plugins/odom.cpp: In member function ‘void mavros::extra_plugins::OdometryPlugin::odom_cb(const ConstPtr&)’:
/home/zein/catkin_pelican/src/mavros/mavros_extras/src/plugins/odom.cpp:234:51: error: ‘struct mavlink::common::msg::ODOMETRY’ has no member named ‘twist_covariance’
   ftf::covariance_urt_to_mavlink(cov_vel_map, msg.twist_covariance);
                                                   ^
make[2]: *** [CMakeFiles/mavros_extras.dir/src/plugins/odom.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/mavros_extras.dir/all] Error 2
make: *** [all] Error 2
cd /home/zein/catkin_pelican/build/mavros_extras; catkin build --get-env mavros_extras | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << mavros_extras:make                    [ Exited with code 2 ]     
Failed    <<< mavros_extras                         [ 47.1 seconds ]           
Abandoned <<< rotors_gazebo                         [ Unrelated job failed ]   
Abandoned <<< test_mavros                           [ Unrelated job failed ]   
_______________________________________________________________________________
Warnings   << rotors_gazebo_plugins:make /home/zein/catkin_pelican/logs/rotors_gazebo_plugins/build.make.000.log

Anybody had similar issue and was able to solve it? 
Thanks!

chanforg, 2020-03-01:
I also have this problem

chanforg, 2020-03-01:
i also have the similar issue

Grindelmort, 2020-03-01:
I changed `twist_covariance` to `velocity_covariance` in **odom.cpp** file and it worked for me.