ethz-asl/rotors_simulator #597: Problem with catkin build

JJavierga, 2020-02-12:
I've followed the instructions given, but, when I get to the point where I have to use catkin build. There are some errors I have no idea where they come from or how to solve them.
I am using ROS indigo in Ubuntu 14.04 and I have gazebo 2.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

These are the lines I think are most important:
Errors     << glog_catkin:make /home/javier/simulation/ros_catkin_ws/logs/glog_catkin/build.make.000.log
...
make[3]: *** [src/logging_unittest-logging_unittest.o] Error 1
make[3]: *** Se espera a que terminen otras tareas.... /*Waiting for other tasks to finish*/
make[2]: *** [glog_src-prefix/src/glog_src-stamp/glog_src-build] Error 2
make[1]: *** [CMakeFiles/glog_src.dir/all] Error 2
make: *** [all] Error 2
...
Failed     << glog_catkin:make                            [ Exited with code 2 ]                  
Failed    <<< glog_catkin                                 [ 32.8 seconds ]                        
Abandoned <<< gazebo_plugins                              [ Unrelated job failed ]                
Abandoned <<< aircraft_plugin                             [ Unrelated job failed ]                
Abandoned <<< gazebo_ros_control                          [ Unrelated job failed ]                
Abandoned <<< liftdrag_plugin                             [ Unrelated job failed ]                
Abandoned <<< mavros_extras                               [ Unrelated job failed ]                
Abandoned <<< rotors_control                              [ Unrelated job failed ]                
Abandoned <<< rotors_description                          [ Unrelated job failed ]                
Abandoned <<< rotors_gazebo_plugins                       [ Unrelated job failed ]                
Abandoned <<< ardupilot_sitl_gazebo_plugin                [ Unrelated job failed ]                
Abandoned <<< rotors_gazebo                               [ Unrelated job failed ]                
Abandoned <<< test_mavros                                 [ Unrelated job failed ]

gsilano, 2020-02-12:
Try to upload to Gazebo 5. Consider that Gazebo 2 has a lot of bugs that have been fixed in the later releases.

JJavierga, 2020-02-12:
In case I change to Gazebo 5, shall I change ROS indigo to any other version? I've seen that mixing some versions of ROS and gazebo might cause some trouble. The main problem here is that I have to use some 5 years old files another person did and I'm not sure wheter they are compatible with newer versions or not.

JJavierga, 2020-02-12:
I've just installed gazebo 7 and I get the same error.

gsilano, 2020-02-12:
It is very strange. I work with it,and everything is fine. Maybe you are following a wrong guide. Did you install all dependencies?

closed 2021-09-25 by Jaeyoung-Lim