ethz-asl/rotors_simulator #598: ERROR Virtual keyboard joystick

marco-marino95, 2020-02-24:
I can't control the quad bike. I followed this guide "Set the joystick of the virtual keyboard" but when I get to the point "rosrun rotors_joy_interface key_joystick.py" the joystick GUI of the virtual keyboard does not exit but gives me this ERROR:

[rosrun] Couldn't find executable named key_joystick.py below /home/marco/catkin_ws/src/CrazyS/rotors_joy_interface
[rosrun] Found the following, but they're either not files,
[rosrun] or not executable:
[rosrun] /home/marco/catkin_ws/src/CrazyS/rotors_joy_interface/scripts/key_joystick.py

What can I do to fix it?

gsilano, 2020-02-25:
Why are you using "CrazyS" instead of "RotorS"? Could you provide the entire verbose?

marco-marino95, 2020-02-26:
I redone all the steps correctly installing rotors, the moment I get to the point: "rosrun rotors_joy_interface key_joystick.py" the GUI of the joystick of the virtual keyboard does not open but gives me this ERROR different from the previous one:

Traceback (most recent call last):
   File "/home/marco/catkin_ws/src/rotors_simulator/rotors_joy_interface/scripts/key_joystick.py", line 9, in <module>
     import pygame, sys, os
ImportError: No module named pygame

gsilano, 2020-02-27:
Could you copy and paste the verbose? It could be missing dependencies or something wrong in the RotorS installation.

marco-marino95, 2020-02-27:
#!/usr/bin/env python
'''
Virtual Joystick from Keyboard
Bharat Tak
September 2016
'''

import uinput, time
import pygame, sys, os
from pygame.locals import *

pygame.init()
BLACK = (0,0,0)
WIDTH = 335
HEIGHT = 406
windowSurface = pygame.display.set_mode((WIDTH, HEIGHT), 0, 32)
windowSurface.fill(BLACK)
pygame.display.set_caption('Virtual RC Joystick')

# Fill background
background = pygame.Surface(windowSurface.get_size())
background = background.convert()
background.fill((250, 250, 250))

# Load image
dir = os.path.dirname(__file__)
filename = os.path.join(dir, '../media/sticks.png')
img = pygame.image.load(filename)

windowSurface.blit(img,(0,0))
pygame.display.flip()

class stick_state(object):
    def __init__(self, name, stick, key_up, key_down, spring_back=True, incr_val=0.2):
        self.name = name                            # The name of the stick
        self.stick = stick                          # The stick on the joystick that this stick maps to 
        self.key_up = key_up                        # The key on the keyboard that maps to this stick increment
        self.key_down = key_down                    # The key on the keyboard that maps to this stick decrement
        self.spring_back = spring_back              # Does the stick spring back to center on release?
        self.incr_val = incr_val                    # The increment on keypress
        self.min_val = 0.0                          # Minimum stick value
        self.max_val = 255.0                        # Maximum stick value
        self.active_up = False                      # True if up key is held pressed
        self.active_down = False                    # True if down key is held pressed
        if self.spring_back:
            self.zero = 127.0
        else:
            self.zero = 0.0
        self.val = self.zero                        # Stick value at initialization at zero position
        self.emit_val = int(self.val)
        self.display_ready = False                  # Whether optional display params have been set
        self.display_height = 0                     # Height on the display screen 
        self.display_width = 0                      # Width on the display screen 
        self.display_hor = True                     # Whether the display bar is horizontal, else vertical
        self.display_bar_g = []
        self.display_bar_b = []

    def keypress_up(self):
        self.active_up = True
        if (self.val + self.incr_val) <= self.max_val:
            self.val = self.val + self.incr_val
        else:
            # Saturated
            self.val = self.max_val

    def keypress_down(self):
        self.active_down = True
        if (self.val - self.incr_val) >= self.min_val:
            self.val = self.val - self.incr_val
        else:
[... 96 lines cut ...]
    roll_stick = stick_state('Roll', uinput.ABS_X, K_RIGHT, K_LEFT)
    roll_stick.set_display(21, 39, True)
    sticks.append(roll_stick)
    pitch_stick = stick_state('Pitch', uinput.ABS_Y, K_UP, K_DOWN)
    pitch_stick.set_display(328, 198, False)
    sticks.append(pitch_stick)
    thr_stick = stick_state('Throttle', uinput.ABS_THROTTLE, K_w, K_s, False)
    thr_stick.set_display(328, 95, False)
    sticks.append(thr_stick)
    rud_stick = stick_state('Yaw', uinput.ABS_RUDDER, K_d, K_a)
    rud_stick.set_display(360, 39, True)
    sticks.append(rud_stick)

    with uinput.Device(events) as device:
        while True:
            # event handling loop
            for event in pygame.event.get():
                for stick in sticks:
                    stick.update_event(event)
            for stick in sticks:
                    stick.update_stick(device)
            time.sleep(0.0005)

if __name__ == "__main__":
    main()

gsilano, 2020-02-27:
No. Maybe you are not familiar with these tools, and I asked you something the wrong way. The verbose is the output that you get on the terminal line when you run the code. I would like to understand if there are errors, missing dependencies, and so on.

marco-marino95, 2020-02-27:
The output that comes out after I run the code is this:

Traceback (most recent call last):
   File "/home/marco/catkin_ws/src/rotors_simulator/rotors_joy_interface/scripts/key_joystick.py", line 9, in
     import pygame, sys, os
ImportError: No module named pygame

gsilano, 2020-02-27:
Did you try to Google it? I found this [question on StackOverflow](https://stackoverflow.com/questions/18317521/importerror-no-module-named-pygame). Please, take a look and let me know.

tooHotSpot, 2020-04-09:
It is just an ```ImportError: No module named pygame```
These would enough to continue the installation:
```pip install pygame```