ethz-asl/rotors_simulator #600: Not able to build all packages

Grindelmort, 2020-03-01:
Hello everyone!
I am an amateur in ROS and trying to work with the **rotors_simulator.** I am following the installation instructions provided at [https://github.com/ethz-asl/rotors_simulator](url) but unfortunately when I run catkin build, some packages fail to build due to some errors.

1. 

> error: ‘const class gazebo::math::Pose’ has no member named ‘Pos’
>    double height_geometric_m = ref_alt_ + model_->GetWorldPose().Pos().Z();
> 

2. 

> error: conversion from ‘const gazebo::math::Pose’ to non-scalar type ‘ignition::math::Pose3d {aka ignition::math::Pose3<double>}’ requested
>    ignition::math::Pose3d T_W_B = link_->GetWorldPose();

3. 

> error: conversion from ‘const gazebo::math::Pose’ to non-scalar type ‘ignition::math::Pose3d {aka ignition::math::Pose3<double>}’ requested
>    ignition::math::Pose3d T_W_I = link_->GetWorldPose();  // TODO(burrimi): Check tf.

4.

>  error: conversion from ‘gazebo::math::Vector3’ to non-scalar type ‘ignition::math::Vector3d {aka ignition::math::Vector3<double>}’ requested
>        link_->GetRelativeLinearAccel() - C_W_I.RotateVectorReverse(gravity_W_);
> 

5.

>  error: conversion from ‘gazebo::math::Vector3’ to non-scalar type ‘ignition::math::Vector3d {aka ignition::math::Vector3<double>}’ requested
>    ignition::math::Vector3d angular_vel_I = link_->GetRelativeAngularVel();
> 

6. 

> error: ‘class gazebo::physics::World’ has no member named ‘GetEntityByName’
>        boost::dynamic_pointer_cast<physics::Link>(world_->GetEntityByName(link_name));

7. 

> error: conversion from ‘gazebo::math::Vector3’ to non-scalar type ‘ignition::math::Vector3d {aka ignition::math::Vector3<double>}’ requested
>    ignition::math::Vector3d W_ground_speed_W_L = link_->GetWorldLinearVel();

I am using Gazebo 7 on Ubuntu 16.04 with ROS Kinetic distro. Can somebody help me with this?

gsilano, 2020-03-01:
Hi @Grindelmort! The problem has been already solved. You find all the instruction in those conversations #592, #590, #599 and #586

Grindelmort, 2020-03-03:
> Hi @Grindelmort! The problem has been already solved. You find all the instruction in these conversations #592, #590, #599 and #586

Thanks for the quick reply. Through the links you mentioned I understood that I need to install BebobS to work my way around this problem. Is there any other way to solve this issue without installing BebobS?

gsilano, 2020-03-03:
You do not need to install BebopS, but the RotorS package that BebopS uses to work.

Grindelmort, 2020-03-04:
Here's what I did after reading the posts you mentioned based on the little I could understand: 
I first installed **BebopS** by following the instructions mentioned at [https://github.com/gsilano/BebopS](this link). There's a **rotors_simulator**  package along with the other files, I copied that package into my workspace having all the Rotors files. Then I ran `catkin build` but it failed.
Then I installed **CrazyS** into the workspace and renamed it to **rotors_simulator**. Again the built failed.
Forgive me if I am acting like a noob.

gsilano, 2020-03-04:
Install `BebopS` as described in the repository. Then, delete all folders related to the Bebop (you have to maintain only `rotors_simulator` and `mav_comm`)

Grindelmort, 2020-03-04:
It worked. Thanks a lot man !!!!

closed 2020-03-04 by Grindelmort

valmil2000, 2021-07-14:
Hey @gsilano once I installed BebopS and removed all packages but mav_comm and rotors_simulator, do you know what the next steps would be to install https://github.com/wilselby/ROS_quadrotor_simulator ? Should I run the catkin build again after pulling the remaining packages or should I do it all in one?

gsilano, 2021-07-14:
HI @valmil2000! I worked with this package a while time ago, but after a while I get stuck. I was fascinated by the use of the OMPL library and the obtained solution for navigation and planning purposes.

The package is based on a very old release of ROS and RotorS. If you take a look at the latest commits, you will see that it's been 4 years. If I'm not mistaken, the package refers to the Indigo Igloo of ROS running on Ubuntu 14.04. This is why you failed in installing the package using the instructions mentioned in this issue. 

I hope this helps. Further information about the package is on [Wilselby's website](https://www.wilselby.com/research/ros-integration/3d-mapping-navigation/).

valmil2000, 2021-07-14:
Thanks for your quick answer @gsilano So you don't think it's compatible?

gsilano, 2021-07-14:
Well, I don't think so. Of course, it depends on how you want to use the package (i.e., RotorS). For instance, if you aim to perform some SLAM, and you don't care about the ROS package to simulate the drone dynamics (no project constraints), you can switch to a more reliable and documented one. The literature is full of solutions to cope with the task.