ethz-asl/rotors_simulator #604: gzserver: symbol lookup error:.../librotors_gazebo_gps_plugin.so: undefined symbol: _ZN14gz_sensor_msgs9NavSatFixC1Ev

070411209, 2020-03-12:
update OS Ubuntu 18.04.4 LTS, and update branch feature/tiltwing, catkin build ok, but roslaunch have error:
[INFO] [1583970473.244816, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[Err] [Plugin.hh:180] Failed to load plugin /home/one/src/rotors_gazebo_plugins/build/librotors_gazebo_ros_interface_plugin.so: /home/one/src/rotors_gazebo_plugins/build/librotors_gazebo_ros_interface_plugin.so: undefined symbol: _ZN21gz_visualization_msgs9VisVector17default_instance_E

...
[INFO] [1583970474.781542, 0.204000]: Spawn status: SpawnModel: Successfully spawned entity
[Err] [Plugin.hh:180] Failed to load plugin /home/one/src/rotors_gazebo_plugins/build/librotors_gazebo_aerodynamics_plugin.so: /home/one/src/rotors_gazebo_plugins/build/librotors_gazebo_aerodynamics_plugin.so: undefined symbol: _ZTIN11gz_std_msgs7Float32E
[Err] [Plugin.hh:180] Failed to load plugin /home/one/src/rotors_gazebo_plugins/build/librotors_gazebo_aerodynamics_plugin.so: /home/one/src/rotors_gazebo_plugins/build/librotors_gazebo_aerodynamics_plugin.so: undefined symbol: _ZTIN11gz_std_msgs7Float32E
[Err] [Plugin.hh:180] Failed to load plugin /home/one/src/rotors_gazebo_plugins/build/librotors_gazebo_aerodynamics_plugin.so: /home/one/src/rotors_gazebo_plugins/build/librotors_gazebo_aerodynamics_plugin.so: undefined symbol: _ZTIN11gz_std_msgs7Float32E
[Err] [Plugin.hh:180] Failed to load plugin /home/one/src/rotors_gazebo_plugins/build/librotors_gazebo_aerodynamics_plugin.so: /home/one/src/rotors_gazebo_plugins/build/librotors_gazebo_aerodynamics_plugin.so: undefined symbol: _ZTIN11gz_std_msgs7Float32E

I have try  OS Ubuntu16.04 is ok

vicmassy, 2020-04-14:
I have the same problem! Any updates on that?

gsilano, 2020-04-15:
Me too. Follow.

hardesh, 2020-05-07:
I'm also getting the same error. I "solved" this once by reinstalling ubuntu. But, the same issue occurred and even after reinstalling Ubuntu I'm getting the same error. 

I am using Ubuntu 18.04.

RapIt, 2020-08-19:
I have the same issue. It turned out that I can run without errors if I am using `catkin_make` instead of `catkin build` from the `python_catkin_tools`. Using ros melodic (ubuntu 18.04). Would be great if this could be resolved.

guiaugustoga, 2020-12-22:
I'm having the same issue with ### librotors_gazebo_multirotor_base_plugin.so ### . Any news on this error ?

Jaeyoung-Lim, 2020-12-22:
@guiaugustoga You should use `catkin build`

guiaugustoga987, 2020-12-23:
> @guiaugustoga You should use `catkin build`

I tried with both catkin_make and catkin build and this error still happens.

hardesh, 2020-12-23:
@guiaugustoga clean Ubuntu installation worked for me 😅

Jaeyoung-Lim, 2021-01-22:
Thanks, @hardesh !

Closing!

closed 2021-01-22 by Jaeyoung-Lim