ethz-asl/rotors_simulator #610: the ROS imu topic can not be found when I use the gazebo_imu_plugin in rotors_gazebo_plugins

unravelwool, 2020-04-06:
Hi, everyone ! I am using the [ardupilot_sitl_gazebo_plugin](https://github.com/AurelienRoy/ardupilot_sitl_gazebo_plugin) to simulate the uav. My uav is Erlecopter which is described in `erlecopter_base.xacro `. This file uses the gazebo_imu_plugin from rotors_gazebo_plugin to produce imu data on the imu ros topic.  The`erlecopter_base.xacro` uses the`<xacro: imu_plugin_macro>` which comes from `component_snippets.xacro` in the following way:
```
<xacro:imu_plugin_macro
    namespace="$(arg nspace)/${namespace}"
    imu_suffix=""
    parent_link="base_link"
    imu_topic="imu_ABCDEFG"
    mass_imu_sensor="0.015"
    gyroscope_noise_density="0.0003394"
    gyroscope_random_walk="0.000038785"
    gyroscope_bias_correlation_time="1000.0"
    gyroscope_turn_on_bias_sigma="0.0087"
    accelerometer_noise_density="0.004"
    accelerometer_random_walk="0.006"
    accelerometer_bias_correlation_time="300.0"
    accelerometer_turn_on_bias_sigma="0.1960"
  >
<inertia ixx="0.00001" ixy="0.0" ixz="0.0" iyy="0.00001" iyz="0.0" izz="0.00001" />
    <origin xyz="0 0 0" rpy="0 0 0" />
  </xacro:imu_plugin_macro>
```
in the above lines, I set imu_topic to "imu_ABCDEFG". And the part about imu in the `component_snippets.xacro` is as follows:
```
<xacro:macro name="imu_plugin_macro"
    params="namespace imu_suffix parent_link imu_topic
      mass_imu_sensor gyroscope_noise_density gyroscope_random_walk
      gyroscope_bias_correlation_time gyroscope_turn_on_bias_sigma
      accelerometer_noise_density accelerometer_random_walk
      accelerometer_bias_correlation_time accelerometer_turn_on_bias_sigma
      *inertia *origin">
    <!-- IMU link -->
    <link name="${namespace}/imu${imu_suffix}_link">
      <inertial>
        <xacro:insert_block name="inertia" />
        <mass value="${mass_imu_sensor}" />  <!-- [kg] -->
        <origin xyz="0 0 0" rpy="0 0 0" />
      </inertial>
    </link>
    <!-- IMU joint -->
    <joint name="${namespace}/imu${imu_suffix}_joint" type="revolute">
      <xacro:insert_block name="origin" />
      <parent link="${parent_link}" />
      <child link="${namespace}/imu${imu_suffix}_link" />
      <limit upper="0" lower="0" effort="0" velocity="0" />
    </joint>
    <gazebo>
      <plugin filename="librotors_gazebo_imu_plugin.so" name="rotors_gazebo_imu${imu_suffix}_plugin">
      <!-- A good description of the IMU parameters can be found in the kalibr documentation:
           https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model-and-Intrinsics -->
        <robotNamespace>${namespace}</robotNamespace> <!-- (string, required): ros namespace in which the messages are published -->
        <linkName>${namespace}/imu${imu_suffix}_link</linkName> <!-- (string, required): name of the body which holds the IMU sensor -->
        <imuTopic>${imu_topic}</imuTopic> <!-- (string): name of the sensor output topic and prefix of service names (defaults to imu) -->
        <gyroscopeNoiseDensity>${gyroscope_noise_density}</gyroscopeNoiseDensity> <!-- Gyroscope noise density (two-sided spectrum) [rad/s/sqrt(Hz)] -->
        <gyroscopeRandomWalk>${gyroscope_random_walk}</gyroscopeRandomWalk> <!-- Gyroscope bias random walk [rad/s/s/sqrt(Hz)] -->
        <gyroscopeBiasCorrelationTime>${gyroscope_bias_correlation_time}</gyroscopeBiasCorrelationTime> <!-- Gyroscope bias correlation time constant [s] -->
        <gyroscopeTurnOnBiasSigma>${gyroscope_turn_on_bias_sigma}</gyroscopeTurnOnBiasSigma> <!-- Gyroscope turn on bias standard deviation [rad/s] -->
        <accelerometerNoiseDensity>${accelerometer_noise_density}</accelerometerNoiseDensity> <!-- Accelerometer noise density (two-sided spectrum) [m/s^2/sqrt(Hz)] -->
        <accelerometerRandomWalk>${accelerometer_random_walk}</accelerometerRandomWalk> <!-- Accelerometer bias random walk. [m/s^2/s/sqrt(Hz)] -->
        <accelerometerBiasCorrelationTime>${accelerometer_bias_correlation_time}</accelerometerBiasCorrelationTime> <!-- Accelerometer bias correlation time constant [s] -->
        <accelerometerTurnOnBiasSigma>${accelerometer_turn_on_bias_sigma}</accelerometerTurnOnBiasSigma> <!-- Accelerometer turn on bias standard deviation [m/s^2] -->
      </plugin>
    </gazebo>
  </xacro:macro>
```
When I launch the “env1.launch” which I write by myself. I can find the imu gazebo topic like that:
```
ubuntu-li@mycomputer:~$ gz topic -l | grep imu
/gazebo/vrc_task_1/launchfile_namespace/erlecopter/imu_ABCDEFG
```
But I do not find the imu ros topic. This may be an error. I think the gazebo_imu_plugin should publish the imu data on an imu ROS topic ,  but the imu ROS topic is missing ! 

Anyone can help me and tell what it happened. I wonder if the gazebo_imu_plugin in rotors_gazebo_plugins can create an imu ros topic and publish imu data to it. 
If the gazebo_imu_plugin can't create an imu ROS topic , what should I do for this?  I find that default sensor plugins in Gazebo are able to create a ROS topic  and publish sensor data to it.
Your help will be greatly appreciated !!

yours sincerely
unravelwool

aeudes, 2020-04-06:
You should check that the world you are using have the ros_interface_plugin.

unravelwool, 2020-04-07:
Thank you so much for your help!! I find that there is no gazebo_ros_interface_plugin in the` .world `file. So I add the following line to my` .world `file:
`<plugin name="ros_interface_plugin" filename="librotors_gazebo_ros_interface_plugin.so"/>`
and run the `env1.launch` file, I find the ros imu topic appears !! It works fine!! 💃:D

But I still have a question. I want simulate several UAVs in the Gazebo. So I  add serveral `erlecopter_base.xacro` into the `<group>` tag in the .launch file. Thus the Gazebo should spawn several erlecopter models, and each erlecopter model has an imu sensor. These imu sensor should create several imu ros topics,  each of which has an different topic name from others. Can only one gazebo_ros_interface_plugin achieve the creation of several imu ros topics ? 
If not, Can I use several gazebo_ros_interface_plugins in a .world file in the following way:
```
<plugin name="ros_interface_plugin_1" filename="librotors_gazebo_ros_interface_plugin.so"/>
<plugin name="ros_interface_plugin_2" filename="librotors_gazebo_ros_interface_plugin.so"/>
<plugin name="ros_interface_plugin_3" filename="librotors_gazebo_ros_interface_plugin.so"/>
......

```
I look forward to hearing from you! :)

unravelwool, 2020-04-09:
with regard to the number of gazebo_ros_interface_plugin in empty.world, one is enough, because the topics which have same functionalities in different uavs are created with different name prefixs. It does not cause any confusion.

closed 2020-04-09 by unravelwool