ethz-asl/rotors_simulator #611: Implementation of angular_rate_error in roll_pitch_yawrate_thrust_controller.cpp does not match with the paper

TaoYibo1866, 2020-04-08:
According to _Control of complex maneuvers for a quadrotor UAV using geometric methods on SE(3)_ [https://arxiv.org/pdf/1003.2005.pdf](url), the anglular rate error should be
`Eigen::Vector3d angular_rate_error = odometry_.angular_velocity - R.transpose() * R_des * angular_rate_des;`.
However, in [https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_control/src/library/roll_pitch_yawrate_thrust_controller.cpp](url) line 111, the implementation is
`Eigen::Vector3d angular_rate_error = odometry_.angular_velocity - R_des.transpose() * R * angular_rate_des;`. Is this a bug?

TaoYibo1866, 2022-07-09:
This cross-multiplication is valid, but always return zero. In this [paper](https://arxiv.org/pdf/1003.2005.pdf), there is

> M = −kR * eR − kΩ * eΩ + Ω × JΩ

The code you mentioned just omitted J.

Scarlett-Sun, 2023-03-18:
I also wonder why the controller for M omits the last part(the long expression) when it is easy to implement?

barrosiuri, 2024-01-10:
Did you guys find out why did they omit the last part of the expression and why they are performing an operation that always returns zero (cross product with itself)?

erichjl, 2025-11-18:
Hi, did you guys figure out why  Eigen::Vector3d angular_rate_error = odometry_.angular_velocity - R_des.transpose() * R * angular_rate_des in the code lee_position_controller.cpp is different from the paper mentioned? I am still curious ...