ethz-asl/rotors_simulator #620: catkin build error [melodic]

sheldonjinqi, 2020-05-15:
I followed the instruction in the readme and when i run catkin build, i got this error. Any suggestions on how to solve this one?
`
Errors     << mavros:make /home/sheldon/catkin_ws/logs/mavros/build.make.000.log
/home/sheldon/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:332:72: error: ‘mavlink::omav’ has not been declared
  void handle_dynamixel(const mavlink::mavlink_message_t *msg, mavlink::omav::msg::DYNAMIXEL_STATUS &dyn_s)
                                                                        ^~~~
/home/sheldon/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:332:100: error: expected ‘,’ or ‘...’ before ‘&’ token
  void handle_dynamixel(const mavlink::mavlink_message_t *msg, mavlink::omav::msg::DYNAMIXEL_STATUS &dyn_s)

/home/sheldon/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp: In member function ‘virtual mavros::plugin::PluginBase::Subscriptions mavros::std_plugins::IMUPlugin::get_subscriptions()’:
/home/sheldon/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:110:52: error: no matching function for call to ‘mavros::std_plugins::IMUPlugin::make_handler(void (mavros::std_plugins::IMUPlugin::*)(const mavlink_message_t*, int))’
           make_handler(&IMUPlugin::handle_dynamixel),
                                                    ^
In file included from /home/sheldon/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:19:0:
/home/sheldon/catkin_ws/src/mavros/mavros/include/mavros/mavros_plugin.h:88:14: note: candidate: template<class _C> mavros::plugin::PluginBase::HandlerInfo mavros::plugin::PluginBase::make_handler(mavlink::msgid_t, void (_C::*)(const mavlink_message_t*, mavconn::Framing))
  HandlerInfo make_handler(const mavlink::msgid_t id, void (_C::*fn)(const mavlink::mavlink_message_t *msg, const mavconn::Framing framing)) {
              ^~~~~~~~~~~~
/home/sheldon/catkin_ws/src/mavros/mavros/include/mavros/mavros_plugin.h:88:14: note:   template argument deduction/substitution failed:
/home/sheldon/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:110:52: note:   candidate expects 2 arguments, 1 provided
           make_handler(&IMUPlugin::handle_dynamixel),
                                                    ^
In file included from /home/sheldon/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:19:0:
/home/sheldon/catkin_ws/src/mavros/mavros/include/mavros/mavros_plugin.h:101:14: note: candidate: template<class _C, class _T> mavros::plugin::PluginBase::HandlerInfo mavros::plugin::PluginBase::make_handler(void (_C::*)(const mavlink_message_t*, _T&))
  HandlerInfo make_handler(void (_C::*fn)(const mavlink::mavlink_message_t*, _T&)) {
              ^~~~~~~~~~~~
/home/sheldon/catkin_ws/src/mavros/mavros/include/mavros/mavros_plugin.h:101:14: note:   template argument deduction/substitution failed:
/home/sheldon/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:110:52: note:   mismatched types ‘_T&’ and ‘int’
           make_handler(&IMUPlugin::handle_dynamixel),
                                                    ^
/home/sheldon/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:111:3: error: could not convert ‘{mavros::plugin::PluginBase::make_handler(void (_C::*)(const mavlink_message_t*, _T&)) [with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::ATTITUDE; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message](&mavros::std_plugins::IMUPlugin::handle_attitude), mavros::plugin::PluginBase::make_handler(void (_C::*)(const mavlink_message_t*, _T&)) [with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::ATTITUDE_QUATERNION; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message](&mavros::std_plugins::IMUPlugin::handle_attitude_quaternion), mavros::plugin::PluginBase::make_handler(void (_C::*)(const mavlink_message_t*, _T&)) [with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::HIGHRES_IMU; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message](&mavros::std_plugins::IMUPlugin::handle_highres_imu), mavros::plugin::PluginBase::make_handler(void (_C::*)(const mavlink_message_t*, _T&)) [with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::RAW_IMU; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message](&mavros::std_plugins::IMUPlugin::handle_raw_imu), mavros::plugin::PluginBase::make_handler(void (_C::*)(const mavlink_message_t*, _T&)) [with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::SCALED_IMU; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message](&mavros::std_plugins::IMUPlugin::handle_scaled_imu), mavros::plugin::PluginBase::make_handler(void (_C::*)(const mavlink_message_t*, _T&)) [with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::SCALED_PRESSURE; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message](&mavros::std_plugins::IMUPlugin::handle_scaled_pressure), <expression error>}’ from ‘<brace-enclosed initializer list>’ to ‘mavros::plugin::PluginBase::Subscriptions {aka std::vector<std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> > >}’
   };
   ^
/home/sheldon/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp: In member function ‘void mavros::std_plugins::IMUPlugin::handle_dynamixel(const mavlink_message_t*, int)’:
/home/sheldon/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:335:61: error: ‘dyn_s’ was not declared in this scope
   dynamixel_status->header.stamp = m_uas->synchronise_stamp(dyn_s.time_boot_us);
                                                             ^~~~~
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/imu.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/mavros_plugins.dir/all] Error 2
make: *** [all] Error 2
`

Lawliet9666, 2020-05-19:
I have the same errors. Have you solved the problem?

gsilano, 2020-05-19:
See #600

sheldonjinqi, 2020-05-19:
> I have the same errors. Have you solved the problem?

I haven’t solved the problem, instead I am using @gsilano ‘  BebopS repo for my simulation. It’s an extension of the RotorS package.

sheldonjinqi, 2020-05-19:
> See #600
Hi gsilano,
One problems I encountered after I installed the BebopS package is that when I run the mav_hovering_example, the robot wouldn’t move. But if I run the test launch file for Bebop, task1_world.launch, it works. Do I have to delete the BebopS folder to get the RotorS package working properly?

gsilano, 2020-05-19:
The problem may be related to the word file. Take a look if something is missing. Compare the one in BebopS with that in the RotorS package.

sheldonjinqi, 2020-05-19:
> The problem may be related to the word file. Take a look if something is missing. Compare the one in BebopS with that in the RotorS package.

Thanks for the quick reply, I will give it a try and post an update here.

sheldonjinqi, 2020-05-19:
Just an update on the solution proposed by @gsilano :
I was able to run the mav_hovering_example using the rotors package that comes with the Bebops repo. The world file was missing the ros interface plugin. After I pasted the following plugin into the world file, everything works.

`        
<plugin name="ros_interface_plugin" filename="librotors_gazebo_ros_interface_plugin.so"/>
`

closed 2020-05-19 by sheldonjinqi

golden-memory, 2020-07-27:
i also meet this problem, can you explain the solution in detail. QAQ

sheldonjinqi, 2020-07-27:
> i also meet this problem, can you explain the solution in detail. QAQ

As I mentioned, I used @gsilano 's Bebop repo for my simulation. The rotors package comes with Bebops repo worked for me. Here is the link to the repo https://github.com/gsilano/BebopS.

golden-memory, 2020-07-30:
> > i also meet this problem, can you explain the solution in detail. QAQ
> 
> As I mentioned, I used @gsilano 's Bebop repo for my simulation. The rotors package comes with Bebops repo worked for me. Here is the link to the repo https://github.com/gsilano/BebopS.

It works great, thank a lot