ethz-asl/rotors_simulator #623: catkin build error:  mavros:make

chuanshuo16nian, 2020-05-25:
Hello,I followed the instruction in the readme ,but I have ROS Molodic on inubuntu 18.04,I got this error and can't slove it. Any suggestions on how to solve this one? thanks a lot.
______________________________________________________________________________
Errors     << mavros:make /home/luxy/catkin_ws/logs/mavros/build.make.002.log  
In file included from /home/luxy/catkin_ws/src/mavros/mavros/src/plugins/setpoint_accel.cpp:20:0:
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h: In member function ‘void mavros::plugin::SetWrenchTargetMixin<D>::set_wrench_target(Eigen::Vector3d, Eigen::Vector3d)’:
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:171:12: error: ‘mavlink::omav’ has not been declared
   mavlink::omav::msg::SET_WRENCH_TARGET_BODY_NED sp;
            ^~~~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:173:3: error: ‘sp’ was not declared in this scope
   sp.force_x = force.x();
   ^~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h: In member function ‘void mavros::plugin::SetTiltAngleTargetMixin<D>::set_tilt_angle_target(float*)’:
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:194:12: error: ‘mavlink::omav’ has not been declared
   mavlink::omav::msg::SET_ROTOR_TILT_TARGET sp;
            ^~~~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:196:4: error: ‘sp’ was not declared in this scope
    sp.alpha[i] = alpha[i];
    ^~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:198:45: error: ‘sp’ was not declared in this scope
   UAS_FCU(m_uas_)->send_message_ignore_drop(sp);
                                             ^~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h: In member function ‘void mavros::plugin::SetTiltrotorActuatorCommandsMixin<D>::set_tiltrotor_actuator_commands(float*)’:
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:212:12: error: ‘mavlink::omav’ has not been declared
   mavlink::omav::msg::SET_TILTROTOR_ACTUATOR_COMMANDS sp;
            ^~~~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:214:4: error: ‘sp’ was not declared in this scope
    sp.u_tiltangles[i] = u[i];
    ^~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:217:4: error: ‘sp’ was not declared in this scope
    sp.u_rotors[i] = static_cast<int>(u[i+6]);
    ^~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:219:45: error: ‘sp’ was not declared in this scope
   UAS_FCU(m_uas_)->send_message_ignore_drop(sp);
                                             ^~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h: In member function ‘void mavros::plugin::SetAttitudeThrustTargetMixin<D>::set_attitude_thrust_target(Eigen::Vector3d, Eigen::Vector3d, Eigen::Quaterniond, Eigen::Vector3d)’:
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:235:12: error: ‘mavlink::omav’ has not been declared
   mavlink::omav::msg::SET_ATTITUDE_THRUST_VECTOR_TARGET_NED sp;
            ^~~~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:236:3: error: ‘sp’ was not declared in this scope
   sp.acc_x = linear_acceleration.x();
   ^~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h: In member function ‘void mavros::plugin::SetAllocationMatrixMixin<D>::set_allocation_matrix(Eigen::VectorXd, Eigen::VectorXd)’:
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:264:12: error: ‘mavlink::omav’ has not been declared
   mavlink::omav::msg::SET_ALLOCATION_MATRIX sp;
            ^~~~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:266:4: error: ‘sp’ was not declared in this scope
    sp.allocation_matrix[i] = allocation_matrix(i);
    ^~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:269:4: error: ‘sp’ was not declared in this scope
    sp.tilt_angles[i] = tilt_angles(i);
    ^~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:271:45: error: ‘sp’ was not declared in this scope
   UAS_FCU(m_uas_)->send_message_ignore_drop(sp);
                                             ^~
In file included from /home/luxy/catkin_ws/src/mavros/mavros/src/plugins/setpoint_velocity.cpp:19:0:
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h: In member function ‘void mavros::plugin::SetWrenchTargetMixin<D>::set_wrench_target(Eigen::Vector3d, Eigen::Vector3d)’:
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:171:12: error: ‘mavlink::omav’ has not been declared
   mavlink::omav::msg::SET_WRENCH_TARGET_BODY_NED sp;
            ^~~~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:173:3: error: ‘sp’ was not declared in this scope
   sp.force_x = force.x();
   ^~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h: In member function ‘void mavros::plugin::SetTiltAngleTargetMixin<D>::set_tilt_angle_target(float*)’:
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:194:12: error: ‘mavlink::omav’ has not been declared
   mavlink::omav::msg::SET_ROTOR_TILT_TARGET sp;
            ^~~~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:196:4: error: ‘sp’ was not declared in this scope
    sp.alpha[i] = alpha[i];
    ^~
/home/luxy/catkin_ws/src/mavros/mavros/include/mavros/setpoint_mixin.h:198:45: error: ‘sp’ was not declared in this scope
[... 272 lines cut ...]
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/setpoint_velocity.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/setpoint_accel.cpp.o] Error 1
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/imu.cpp.o] Error 1
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/setpoint_trajectory.cpp.o] Error 1
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/setpoint_raw.cpp.o] Error 1
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/setpoint_position.cpp.o] Error 1
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/setpoint_attitude.cpp.o] Error 1
make[1]: *** [CMakeFiles/mavros_plugins.dir/all] Error 2
make: *** [all] Error 2
cd /home/luxy/catkin_ws/build/mavros; catkin build --get-env mavros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << mavros:make                           [ Exited with code 2 ]     
Failed    <<< mavros                                [ 10.8 seconds ]           
Abandoned <<< mavros_extras                         [ Unrelated job failed ]   
Abandoned <<< rotors_gazebo_plugins                 [ Unrelated job failed ]   
Abandoned <<< rotors_gazebo                         [ Unrelated job failed ]   
Abandoned <<< rotors_hil_interface                  [ Unrelated job failed ]   
Abandoned <<< test_mavros                           [ Unrelated job failed ]   
[build] Summary: 13 of 19 packages succeeded.                                  
[build]   Ignored:   2 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 5 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 13.9 seconds total.

wrld, 2020-05-29:
I have the same problem, but I use kinetic, have you already solved it?

csyang6052, 2020-06-08:
I have the same issue with Ubuntu 18.04 and ROS 'melodic'.
~/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:335:61: error: 'dyn_s' was not declared in this scope

csyang6052, 2020-06-12:
The issue mentioned above was solved using "BebopS", the link is https://github.com/gsilano/BebopS

mei123hao, 2020-07-16:
> The issue mentioned above was solved using "BebopS", the link is https://github.com/gsilano/BebopS

have you installed beboops sucessful on ubuntu18.04？I always got  error like following when catkin build ：
Failed     << bebop_driver:make                    [ Exited with code 2 ]                                                                      
Failed    <<< bebop_driver                         [ 17.0 seconds ]                                                                            
Abandoned <<< bebop_tools                          [ Unrelated job failed ]                                                                    
[build] Summary: 3 of 5 packages succeeded.                                                                                                    
[build]   Ignored:   1 packages were skipped or are blacklisted.                                                                               
[build]   Warnings:  1 packages succeeded with warnings.                                                                                       
[build]   Abandoned: 1 packages were abandoned.                                                                                                
[build]   Failed:    1 packages failed.                                                                                                        
[build] Runtime: 28.1 seconds total.                                                                                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

Please ！ Help！

maxb91, 2020-08-14:
The first issue should be fixed by #636, you'll need to remove your local clone of mavros and install the official mavros instead:

>  sudo apt-get install ros-kinetic-mavros

or for melodic:
>  sudo apt-get install ros-melodic-mavros

closed 2020-08-14 by maxb91