ethz-asl/rotors_simulator #624: catkin build error: mavros make

Gallifery, 2020-05-27:
The Ubuntu version is 16.04.6
---------------------------------------
[image]
[image]

ChannalYang, 2020-06-03:
I've got the same error, have you ever solved it?

Jinghao-coding, 2020-06-09:
I've got the same error, have you ever solved it?

OreoWangH, 2020-06-23:
I've got the same error, have you ever solved it?

ChannalYang, 2020-06-23:
I solved the problem by transferring to “BebopS” . You can find the repository in github also.

May

OreoWangH, 2020-06-25:
> I solved the problem by transferring to “BebopS” . You can find the repository in github also. May
> […](#)

hi, thanks for your reply! But I didn't solve this problem. I found BebopS in [https://github.com/gsilano/BebopS] and copy the otors_simulator and mav_comm by 
git clone -b med18 https://github.com/gsilano/rotors_simulator.git 
git clone -b med18 https://github.com/gsilano/mav_comm.git
Then I build the project again and get the same error. I was confused that why it needed BebopS when the error was in the package of mavros.

Leekxwj, 2020-07-03:
I have met the same problem, have you figured it out?

golden-memory, 2020-07-28:
I meet the same problem

RainySun-GT, 2020-09-03:
I have met the same problem, how to fix it?

rikba, 2020-09-03:
Hi 

We thought we fixed this in https://github.com/ethz-asl/rotors_simulator/pull/636/files. Can you try removing git: {local-name: mavros, uri: 'https://github.com/ethz-asl/mavros.git'} from your catkin workspace and install `sudo apt-get install ros-kinetic-mavros` instead?

RainySun-GT, 2020-09-03:
> Hi
>
> We thought we fixed this in https://github.com/ethz-asl/rotors_simulator/pull/636/files. Can you try removing git: {local-name: mavros, uri: [...]

So there is no need of a file folder named "mavros" in the "src" file folder in my catkin workspace? But it seems that it still reports errors. Here are the screenshot.

[image]

sunnyshi0310, 2020-09-20:
> So there is no need of a file folder named "mavros" in the "src" file folder in my catkin workspace? But it seems that it still reports errors. Here are the screenshot.
>
> [image]

I meet the same problem...

simao2012, 2021-01-12:
Try this, it should solve the problem :
$ cd catkin_ws
$ rosdep install --from-paths src -i