ethz-asl/rotors_simulator #626: MAV rotors don not spin

Bibbidi-Babbidi-Boo, 2020-06-15:
Hey, so I followed all the steps given here. I was able to build successfully on Ros Melodic Gazebo 9. I am able to spawn a world with the MAV. However when I give the rotor speed commands, the rotors do not spin. Any help would be great

Thanks

mbdilaver, 2020-06-17:
Same problem here. This package was working fine on ROS Melodic about 6 months ago. I've decided to make a fresh installation of ROS Melodic on a Ubuntu 18.04 machine 3 months ago. It's not been working since then. I really wonder how something working became broken and not fixed for a long time. It's sad.

mbdilaver, 2020-06-17:
Hi! I've found a temporary solution. Go to `rotors_simulator/rotors_gazebo_plugins/CMakeLists.txt` and add `list(APPEND targets_to_install mav_msgs)` line after the line `target_link_libraries(mav_msgs ${PROTOBUF_LIBRARY} gazebo_msgs)`. 

```cmake
...
add_library(mav_msgs SHARED ${PROTO_SRCS})
target_link_libraries(mav_msgs ${PROTOBUF_LIBRARY} gazebo_msgs)

list(APPEND targets_to_install mav_msgs) # <----- NEW LINE

# This causes mav_msgs to be linked with every created library in this file from this
# point forward.
# NOTE: This is deprecated, should be using target_link_libraries instead
link_libraries(mav_msgs)
...
```
Go to your workspace root and run the commands below.
```bash
catkin config --install
catkin build
```
Then go to `.bashrc` and add source line for `install` directory of your workspace, not `devel`. Like this:
```
source <your_workspace>/install/setup.bash
```
When I run `hovering_example.launch` it now works and flies.

```
roslaunch rotors_gazebo mav_hovering_example.launch 
````

mbdilaver, 2020-06-17:
Also you can use `catkin_make` instead of `catkin build` and source your `devel/setup.bash`. Sourcing `install` directory may require additional work on `CMakeLists.txt` of other packages you have. After I compile with `catkin_make` drones fly without a problem.

Reference:
[https://github.com/gsilano/CrazyS/issues/40#issuecomment-615197724](https://github.com/gsilano/CrazyS/issues/40#issuecomment-615197724)

Bibbidi-Babbidi-Boo, 2020-06-17:
Yeah its sad that some change causes so much problems.

Anyway, catkin_make didnt work for me. I used catkin_make_isolated, which compiled completely, but gave same non flying results. But for some reason, uninstalling and re-installing ROS made the simulation work. I havent checked fully, so still not entirely sure. Thank for this too, I'll try it out

aka-LJ, 2020-08-05:
> Hi! I've found a temporary solution. Go to `rotors_simulator/rotors_gazebo_plugins/CMakeLists.txt` and add `list(APPEND targets_to_install mav_msgs)` line after the line `target_link_libraries(mav_msgs ${PROTOBUF_LIBRARY} gazebo_msgs)`.
> 
> ```cmake
> ...
> add_library(mav_msgs SHARED ${PROTO_SRCS})
> target_link_libraries(mav_msgs ${PROTOBUF_LIBRARY} gazebo_msgs)
> 
> list(APPEND targets_to_install mav_msgs) # <----- NEW LINE
> 
> # This causes mav_msgs to be linked with every created library in this file from this
> # point forward.
> # NOTE: This is deprecated, should be using target_link_libraries instead
> link_libraries(mav_msgs)
> ...
> ```
> 
> Go to your workspace root and run the commands below.
> 
> ```shell
> catkin config --install
> catkin build
> ```
> 
> Then go to `.bashrc` and add a source line for `install` directory of your workspace instead of `devel` like below.
> 
> ```
> source <your_workspace>/install/setup.bash
> ```
> 
> When I run `hovering_example.launch` it now works and flies.
> 
> ```
> roslaunch rotors_gazebo mav_hovering_example.launch 
> ```
> 
> References:
> [#506 (comment)](https://github.com/ethz-asl/rotors_simulator/pull/506#issuecomment-515064729)
> [gsilano/CrazyS@86d80d3](https://github.com/gsilano/CrazyS/commit/86d80d318c8b02f465e778b17677766422470bda)

Hi, I've tried this, but getting the error message, any idea which CMakeList.text I should work on? thank you. 

ERROR: cannot launch node of type [rotors_gazebo/waypoint_publisher]: Cannot locate node of type [waypoint_publisher] in package [rotors_gazebo]. Make sure file exists in package path and permission is set to executable (chmod +x)

ERROR: cannot launch node of type [rotors_control/lee_position_controller_node]: Cannot locate node of type [lee_position_controller_node] in package [rotors_control]. Make sure file exists in package path and permission is set to executable (chmod +x)

aka-LJ, 2020-08-05:
> Hi, I've tried this, but getting the error message, any idea which CMakeList.text I should work on? thank you.
>
> ERROR: cannot launch node of type [rotors_gazebo/waypoint_publisher]: Cannot locate [...]

This is really strange, I manage to get the mav hovering example to work by building the CrazyS pack instead of the BehopS package without any change.