ethz-asl/rotors_simulator #629: catkin build error in ubuntu18.04

wbzhang233, 2020-06-28:
Hey, I'm facing an error below. My OS is ubuntu18.04, and I obey all the instructions, but I got failed.
`

Errors     << rotors_gazebo_plugins:cmake /home/wbzhang/catkin_obs/logs/rotors_gazebo_plugins/build.cmake.001.log
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindBoost.cmake:911 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "chrono" will no longer be dereferenced when the
  policy is set to NEW.  Since the policy is not set the OLD behavior will be
  used.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:1558 (_Boost_MISSING_DEPENDENCIES)
  /usr/share/OGRE/cmake/modules/FindOGRE.cmake:318 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:175 (find_package)
  /opt/ros/melodic/share/gazebo_dev/cmake/gazebo_dev-extras.cmake:2 (find_package)
  /opt/ros/melodic/share/gazebo_dev/cmake/gazebo_devConfig.cmake:222 (include)
  CMakeLists.txt:125 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /home/wbzhang/catkin_obs/src/rotors_simulator/rotors_gazebo_plugins/cmake/FindGlog.cmake:77 (MESSAGE):
  Failed to find glog - Could not find glog include directory, set
  GLOG_INCLUDE_DIR to directory containing glog/logging.h
Call Stack (most recent call first):
  cmake/FindGlog.cmake:103 (GLOG_REPORT_NOT_FOUND)
  CMakeLists.txt:147 (find_package)

`

weblucas, 2020-06-29:
you have to install the dependencies ...
either manually or via one of the rosinstall file in the base of the repo such as dependencies.rosinstall

JamalShams, 2020-11-10:
You can use cmake_yourworkstation_isolated it worked for me....

closed 2021-01-19 by Jaeyoung-Lim