ethz-asl/rotors_simulator #633: Controlling a single payload with multiple drones?

maxcapobianco, 2020-07-22:
Hi, 

I am trying to control a hanging payload with multiple drones. I currently spawn two drones in separate namespaces, each being controlled independently. I have created a URDF for the payload and can easily have it slung under a single drone, is it possible to link the two separate drones to the one payload although they are in separate namespaces? If so how would you recommend achieving this? 

I have seen something similar done in this paper https://journals.sagepub.com/doi/abs/10.1177/0278364919854131

Thanks!

gsilano, 2020-07-22:
I don't think they made available as open-source this simulation scenario. I think that the best way is to send an email to the authors asking for this. Of course, if you got a positive answer, do not forget to upload this issue.

maxcapobianco, 2020-07-22:
Thanks for the quick response, I'll try to get in contact with the authors. Would you recommend any other search terms that I could use to find a workaround for linking two separate robots in Gazebo?

gsilano, 2020-07-22:
If you are referring to Tagliabue's paper, I think the only way to deal with the problem is to build your own model in Gazebo. This should not be so much difficult. If you are familiar with joints, e.g., revolute, fixed, (see [this link](http://gazebosim.org/blog?page=2)), in a few steps you should succeed in this task. I suggest taking a look in RotorS, maybe there is something already available with the [Voliro platform](https://www.youtube.com/watch?v=Sgd0i87r9-0).

maxcapobianco, 2020-07-24:
I have pretty much got a working model with two drones flying cooperatively with individual payloads. The main issue I am having is defining a payload that is joined between the two drones. The two drones are setup in separate namespaces, where would I define the payload?

gsilano, 2020-07-24:
You have to build your own xacro or sdf file to model the whole scenario. Take a look at [this folder](https://github.com/ethz-asl/rotors_simulator/tree/master/rotors_gazebo/models) to somehow understand how to deal with the task.

maxcapobianco, 2020-07-30:
Found a guide on setting up a collaborative lift situation [here](https://core.ac.uk/download/pdf/83464273.pdf).

closed 2020-07-30 by maxcapobianco