ethz-asl/rotors_simulator #634: Getting a controller, that assumes a cross configuration, working with a quadrotor that assumes plus configuration physics

JeffR1992, 2020-07-22:
I've written an LQI controller that works well for the Hummingbird quadrotor in its default plus prop configuration (as seen below):
[image]

However, I wanted to try and get the Hummingbird to instead fly in a cross prop configuration, and thus changed the controller to assume such a configuration (as seen below):
[image]

But, when trying the modified controller on the Hummingbird, it fails to achieve even a stable hover. I'm assuming this is because the dynamic equations used in the physics for the Hummingbird assume a plus prop config, and so trying to control it using a controller that assumes a cross prop config causes an inconsistency between the controller and the physics. As such, I was wondering if there's a way to switch the Hummingbird's physics to also assume a cross prop config?

gsilano, 2020-07-23:
By modifying [this file](https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_description/urdf/hummingbird.xacro), in particular line 87, 108, and so on, you can change the position of the rotors in the 3D space. This is what do you need to change the shape of the quad-rotor, from a + to an X configuration.

Stavya1993, 2020-11-13:
@gsilano I changed the lines mentioned in the xacro files, but that did not change the config. of my quad. I also took the multirotor_base file and changes the joint wrt base_link_inertia and connected all my rotors to base_link_inertia. This changed the configuration to a x config, but i suppose the default controller doesn't work well with x config.