ethz-asl/rotors_simulator #641: Build Fail

a-tab-a, 2020-09-24:
HI,

I have previously successfully built this simulator,however I was trying to install it in a friend's computer. I am getting following error:

Errors     << rotors_gazebo_plugins:make /home/ryans/catkin_ws/logs/rotors_gazebo_plugins/build.make.001.log
In file included from /home/ryans/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_mavlink_interface.cpp:22:0:
/home/ryans/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins/include/rotors_gazebo_plugins/gazebo_mavlink_interface.h:62:10: fatal error: mavlink/v2.0/common/mavlink.h: No such file or directory
 #include <mavlink/v2.0/common/mavlink.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

I built mavros from original repository and whenever I was trying to build the simulator, it shows the error. I have mavcomm repo for your git. My system is
Ubuntu 18.04, Ros-Melodic-Gazebo 9
I assume the header file location is mavlink build folder? I could see the header file in there, not sure why it says no file in directory. Any idea/solution?

patripfr, 2020-11-16:
I'm having the same issue, also on Ubuntu 18.04 with ros melodic and Gazebo 9

l-schilling, 2021-01-23:
Are you sure git was able to download everything correctly? I had an error using git clone and needed to download the zip directly from github. Then it worked.

closed 2021-09-25 by Jaeyoung-Lim