ethz-asl/rotors_simulator #642: How to install it in Ubuntu20.04 with ROS Noetic

thomassmile, 2020-09-27:
Hi, I have Ubuntu 20.04 with ROS Noetic. How can I install rotors_simulator?

BTW, Ubuntu install of ROS Noetic is shown in the website: http://wiki.ros.org/noetic/Installation/Ubuntu

thomassmile, 2020-09-28:
It is similar to that of Ubuntu 16.04 with ROS Kinetic

closed 2020-09-28 by thomassmile

fmascarich, 2021-03-15:
There were some changes I had to make because of OpenCV:

rotors_gazebo_plugins/src/external/gazebo_geotagged_images_plugin.cpp
change #Include <cv.h> to #include <opencv2/opencv.hpp>
change #Include <highgui.h> to #include #include <opencv2/highgui/highgui.hpp>
line 145: cv::COLOR_BGR2RGB

rotors_gazebo_plugins/src/gazebo_odometry_plugin.cpp:
line 92: covariance_image_ = cv::imread(image_name, cv::IMREAD_GRAYSCALE);