ethz-asl/rotors_simulator #646: How to attach Lidar on the model of FireFly

taminhtoan2601, 2020-10-14:
Hello all, 

I am newbie to Gazebo and ROS. Now, I want to attach Lidar sensor to the model Firefly to get data in order to control the firefly. With the simple models I created, I can attach it easily but with the ros_simulator package I downloaded I don't know where the file urdf in the package and in that file urdf where I can add the commands to attach it. How can I do it? Could you explain me little detail? Thank you very much for helping me!

Have a good day ahead!

hmf97, 2020-11-09:
You need to go to rotors_description file, add the xacro file of the sensor you want to use. Then open the "firefly_base.xacro" file and include your xacro sensor file and then your sensor.
The built-in sensors are found in "component_snippets.xacro" if you wish to see an example.

talhamagat, 2024-02-07:
Hello, should I also add a urdf file while doing all this? Will only firefly.xacro and firefly_base.xacro files be changed?