ethz-asl/rotors_simulator #650: Using external position estimator in Rotors_gazebo

Duckyu, 2020-12-03:
Hi, I'm trying to compare my algorithm with the SOTA methods implemented on the rotors_gazebo simulator. To evaluate my algorithm, I need to use the external position estimator for UAV flight(not using the simulator's GT position). I've already successfully used external position estimator with mavros package and PX4 SITL simulator(with rostopic /mavros/vision_pose/pose). I wonder if there is a way to apply the external position estimation while I'm using the rotors_gazebo simulator as the PX4 SITL simulatordoes. Please let me know if anyone have experiences related to this work. Thank you!!

Jaeyoung-Lim, 2020-12-07:
@Duckyu Could you clarify on what you are trying to achieve? What component are you trying to use from rotorS and what from PX4/sitl_gazebo?

Duckyu, 2020-12-07:
@Jaeyoung-Lim 
 Sorry for my unclear question.
 I want to test my own algorithm on the rotorS gazebo simulator. In my algorithm, UAV has to be controlled by the estimated position data(such as the result of the VINS-fusion VI-O algorithm). Therefore, I am looking for the method to put external position estimation result as input odometry data of the controller while using the rotors simulator.
 Thank you for replying to me.

Jaeyoung-Lim, 2020-12-07:
@Duckyu Okay, but you can already do all of that with sitl_gazebo. My question was what your motivations were with moving over to rotorS?

engcang, 2020-12-07:
@Jaeyoung-Lim Hi, Jaeyoung.
I am also in the same situation with @Duckyu.
In my case, I want to test and compare my own UAV algorithm with SOTA algorithms, which mostly use rotorS basically.
They mainly focus on exploring the unknown environments as fast as possible without considering the pose estimation, whereas I am trying to consider pose estimation too.

Jaeyoung-Lim, 2020-12-07:
@engcang But the only differences are in the models and the plugins. Is it just because you don't want to modify the launchfiles of rotorS? I still don't understand what the problem is

engcang, 2020-12-07:
@Jaeyoung-Lim This is because I do not want to modify the original SOTA algorithms, I mean topic names, services, flag variables, controller, and anything else.
So I just want to run their algorithms with rotorS and also with external pose estimation.
I don't need PX4-SITL if I can do so.
But if not, then I would eventually have to modify the algorithms to run it with PX4 SITL.

Jaeyoung-Lim, 2020-12-08:
@engcang Okay, but rotors doesn't have a position estimator. Where do you plan to run your estimator to fuse the vision estimate if you don't run it with SITL?

closed 2020-12-08 by Jaeyoung-Lim