ethz-asl/rotors_simulator #655: undefined symbol: _ZN14gz_sensor_msgs9ActuatorsC1Ev

YunwooLee94, 2021-01-15:
Hello,
I am working on ROS melodic (ubuntu18.04).
When I installed rotorS and launched "mav_hovering_example.launch", I met this error message.

gzserver: symbol lookup error: /opt/ros/melodic/share/gazebo_plugins/../../lib/librotors_gazebo_multirotor_base_plugin.so: undefined symbol: _ZN14gz_sensor_msgs9ActuatorsC1Ev

Some issues seem similar, but their tries are ineffective in my case.
Is there any solution exists with this problem?

Jaeyoung-Lim, 2021-01-19:
@YunwooLee94 Have you tried rebuilding your workspace after a `catkin clean` ?

YunwooLee94, 2021-01-20:
@Jaeyoung-Lim 
I solved the problem.
My try was rebuilding my workspace then installed previous version of RotorS.
Thank you.

closed 2021-01-20 by YunwooLee94

SunggooJung, 2021-03-02:
@YunwooLee94 
Hi, which version of RotorS did you install?
catkin clean and rebuilding is not works for me

Jaeyoung-Lim, 2021-03-02:
@SunggooJung Please post a new issue with a detailed log of your issue.

It is most likely that your issue is not related to this one