ethz-asl/rotors_simulator #657: rotors_hil.rosinstall Does not install the rotors_simulator package

00nick14, 2021-01-26:
I installed the rotors_simulator using the commands
[image]
but the merge command only added mavlink and mavros folders to my workspace. A friend with a working rotor_simulator package had the line
`- git: {local-name: rotors_simulator, uri: 'https://github.com/ethz-asl/rotors_simulator.git'`
in his `rotors_hil.rosinstall` file, and when I added this line and used `wstool merge rotors_hil.rosinstall` everything worked fine.

Jaeyoung-Lim, 2021-01-26:
@00nick14 It is closer to standard practice that you clone this repo first and then use the rosinstall instructions inside it to pull in the dependencies, compared to what you are doing by getting the rosinstall file using `wget`. That is why the self reference was removed in https://github.com/ethz-asl/rotors_simulator/pull/651

Would this be a sufficient explanation for the change?

00nick14, 2021-01-26:
Yes, although I think the install instructions should include cloning the repo if that's a critical step. I'm new to Linux and ROS so I'm not really aware of the standard practices, and was just following the documentation.

closed 2021-09-25 by Jaeyoung-Lim