ethz-asl/rotors_simulator #661: undefined symbol: _ZN14gz_sensor_msgs9ActuatorsC1Ev

SunggooJung, 2021-03-02:
I'm trying to use RotorS for system ID purpose but it show the errors like below.
Have any idea?
----------------------------------------------------------------------------------------------------------------- 
$ roslaunch rotors_gazebo mav.launch
... logging to /home/sg/.ros/log/a06429c2-7b24-11eb-aaac-a85e45e6c65b/roslaunch-sg-10146.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global property: pi
when processing file: /home/sg/catkin_ws/src/rotors_simulator/rotors_description/urdf/component_snippets.xacro
included from: /home/sg/catkin_ws/src/rotors_simulator/rotors_description/urdf/firefly_base.xacro
started roslaunch server http://sg:36581/

SUMMARY
========

PARAMETERS
 * /firefly/robot_description: <?xml version="1....
 * /firefly/tf_prefix: 
 * /gazebo/enable_ros_network: True
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
  /firefly/
    spawn_firefly (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [10159]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a06429c2-7b24-11eb-aaac-a85e45e6c65b
process[rosout-1]: started with pid [10170]
started core service [/rosout]
process[gazebo-2]: started with pid [10173]
process[gazebo_gui-3]: started with pid [10182]
process[firefly/spawn_firefly-4]: started with pid [10187]
[ INFO] [1614668281.515324299]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1614668281.516312139]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1614668281.542289940]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1614668281.543125175]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[INFO] [1614668281.765485, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1614668281.770620, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[ INFO] [1614668282.815874700]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1614668282.827225844]: Physics dynamic reconfigure ready.
[INFO] [1614668282.974797, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1614668283.090899, 0.000000]: Spawn status: SpawnModel: Successfully spawned entity
gzserver: symbol lookup error: /home/sg/catkin_ws/devel/lib/librotors_gazebo_multirotor_base_plugin.so: undefined symbol: _ZN14gz_sensor_msgs9ActuatorsC1Ev
[gazebo-2] process has died [pid 10173, exit code 127, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -u -e ode /home/sg/catkin_ws/src/rotors_simulator/rotors_gazebo/worlds/basic.world __name:=gazebo __log:=/home/sg/.ros/log/a06429c2-7b24-11eb-aaac-a85e45e6c65b/gazebo-2.log].
log file: /home/sg/.ros/log/a06429c2-7b24-11eb-aaac-a85e45e6c65b/gazebo-2*.log
[firefly/spawn_firefly-4] process has finished cleanly
log file: /home/sg/.ros/log/a06429c2-7b24-11eb-aaac-a85e45e6c65b/firefly-spawn_firefly-4*.log
================================================================================REQUIRED process [master] has died!
process has died [pid 10159, exit code -9, cmd rosmaster --core -p 11311 -w 3 __log:=/home/sg/.ros/log/a06429c2-7b24-11eb-aaac-a85e45e6c65b/master.log].
log file: /home/sg/.ros/log/a06429c2-7b24-11eb-aaac-a85e45e6c65b/master*.log
Initiating shutdown!
================================================================================
[gazebo_gui-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

SunggooJung, 2021-03-03:
I solved this by rebuilding with force cmake. 
$ catkin build --force-cmake

jlack1987, 2021-04-09:
I have the same issue, is there a workaround or anything?

zyhphd, 2021-12-05:
I found the reason. You just need to copy the Actuators.pb.cc and Actuators.pb.h to rotors_gazebo_plugins, and add it to your cmakelist.txt where you generate the librotors_gazebo_multirotor_base_plugin.so.   
add_library(rotors_gazebo_multirotor_base_plugin SHARED src/gazebo_multirotor_base_plugin.cpp src/Actuators.pb.cc)

SunggooJung, 2022-01-09:
> I have the same issue, is there a workaround or anything?

closed 2022-01-09 by SunggooJung

jlack1987, 2022-01-10:
I fixed it and MR'd the fix but it hasn't been merged yet. Probably shouldn't close until the MR has been merged
https://github.com/ethz-asl/rotors_simulator/pull/667

xiaoshengjianan, 2023-04-27:
How did you fixed it? @SunggooJung

jlack1987, 2023-04-27:
I fixed it in that MR but it was never merged. Somebody needs to merge [this](https://github.com/ethz-asl/rotors_simulator/pull/667) to fix. Been awhile now, I suspect this code is not maintained anymore