ethz-asl/rotors_simulator #668: catkin build error when installing RotorS

wendy-ziwen-meng, 2021-05-29:
When installing RotorS following the instructions for 16.04 kinetic on https://github.com/ttdm/rotors_simulator,
I faced the error when typed in the 'catkin build', the errors are shown below:
[image]
[image]
[image]
[image]
What should I do to fix this?

Jaeyoung-Lim, 2021-05-29:
You can try installing the missing python dependencies.

However, ROS Kinetic is End of Life since April, therefore wont be maintained further on

tturkmen, 2021-08-08:
I use ros melodic. Thanks. I will format computer than try again.

Jaeyoung-Lim, 2021-09-25:
Closing as stale

closed 2021-09-25 by Jaeyoung-Lim