ethz-asl/rotors_simulator #669: Symbol Lookup Error: librotors_gazebo_multirotor_base_plugin.so

hariharanr1799, 2021-06-12:
Error while running
`roslaunch rotors_gazebo mav_hovering_example.launch mav_name:=firefly world_name:=basic
`
Error:
`gzserver: symbol lookup error: /home/hariharan/eth_ws/devel/lib/librotors_gazebo_multirotor_base_plugin.so: undefined symbol: _ZN14gz_sensor_msgs9ActuatorsC1Ev`

I'm using ROS Melodic on Ubuntu 18.04. Have tried solutions from other issues or sources.
Reinstalled ROS and Gazebo. Did `catkin clean` and then built again. Nothing works.

Is there any solution to rectify this issue?

zyhphd, 2021-12-02:
Hi, i met the same error, do you find the solution? Looking forward to your reply.

zyhphd, 2021-12-05:
I found the reason. You just need to copy the Actuators.pb.cc and Actuators.pb.h to rotors_gazebo_plugins, and add it to your cmakelist.txt where you generate the librotors_gazebo_multirotor_base_plugin.so.
add_library(rotors_gazebo_multirotor_base_plugin SHARED src/gazebo_multirotor_base_plugin.cpp src/Actuators.pb.cc)

ana-lys, 2022-03-20:
I also got this problem. I am using Ros Noetic and Gazebo 11.3. I tried both the rotors_simulator that come with flightmare and the  rotors_simulator master one. The mentioned method above is not working for me.