ethz-asl/rotors_simulator #670: Keyboard Control Failure

natunator, 2021-06-16:
Hello, I am trying to simulate rotors in a gazebo environment. I am trying manual ( keyboard ) control for now, however I don't seem to understand the Thrust-roll-pitch-yaw controls in the joystick. The throttle and yaw do not work, i.e. they are non-responsive. Pitch seems to tilt, but also elevate ( gain altitude ) the drone. Only roll is working fine. What can I do to fix this ?
Also, when I launch the gazebo launch file, i get these two errors:

[ERROR] [1623858667.176245245]: Couldn't open joystick /dev/input/js0. Will retry every second.

[ WARN] [1623858668.223723209]: Couldn't set gain on joystick force feedback: Bad file descriptor

I checked /dev/input/. There are several event files, as well as 'mice' files, but no js files

closed 2021-06-17 by natunator

Pondigo, 2022-05-07:
Hello, how resolve this?

minisungjun, 2025-03-04:
for me on Ubuntu 20.04 with noetic:

1. first run
$ rosrun rotors_joy_interface key_joystick.py
-> then if u run $ ls /dev/input/, you'll see jsX (for me it is js0)

2. add the following line in mav_with_keyboard.launch in the <group ns>
<rosparam param="joy_node/dev">/dev/input/js0</rosparam>

3. then run
$ roslaunch rotors_gazebo mav_with_keyboard.launch

Now I you can control the drone with the key board, but it is difficult to control the drone