ethz-asl/rotors_simulator #673: Build in Noetic

a-tab-a, 2021-07-16:
Hello,

I am trying to build this Ubuntu 20.04 ROS Noetic but could not do so and getting following error.  It is related to ignition-common3-graphics, I install this package along with gazebo and gazebo does start up. It shows there is a version mismatch, how to work around this? Thanks
[image]
[image]

a-tab-a, 2021-07-16:
Never mind just upgrading works! Closing since got it work.

closed 2021-07-16 by a-tab-a

zhengleily, 2021-09-28:
I encountered the same issue” . How do you fix this error?

zhengleily, 2021-09-28:
I encountered the same issue!
How do you fix this error exactly?