ethz-asl/rotors_simulator #680: How to use the .rosinstall files to install the dependencies?

SchandraS, 2021-09-30:
I'm confused on how to use the rosinstall files to install the dependencies.

Jaeyoung-Lim, 2021-09-30:
@SchandraS I believe your question is not directly related to this repository, and more of a question related to ROS

You can look up the tutorial in documentations like https://docs.ros.org/en/independent/api/rosinstall/html/

closed 2021-09-30 by Jaeyoung-Lim

tkazik, 2021-10-01:
@SchandraS You can for instance use [vcstool](https://github.com/dirk-thomas/vcstool) and invoke the following command in the `src` directory of your workspace:

```shell
vcs import < rotors_simulator/rotors_minimal.rosinstall
```

This will fetch all the packages required to build rotors. 

Note1: In the case of [rotors_minimal.rosinstall](https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_minimal.rosinstall) it is only one package, namely [mav_comm](https://github.com/ethz-asl/mav_comm)
Note2: The [dependencies.rosinstall](https://github.com/ethz-asl/rotors_simulator/blob/master/dependencies.rosinstall) seems to pull in some outdated/broken packages (at least for Ubuntu20.04)...we might need to fix these.