ethz-asl/rotors_simulator #694: Unstable drone on faster than real time simulations

markostamos, 2022-01-25:
Drone gets unstable and falls when update rate of gazebo goes above 2x real time.

Jaeyoung-Lim, 2022-01-25:
@markostamos I think this issue is lacking context / details to decide whether if this is a real issue.

How are you controlling the vehicle? Could you provide instruction can we reproduce your problem?

markostamos, 2022-01-25:
i set a position (x,y,z)  using the lee controller. As expected this works and the uav stays in that position but if i try to do the same thing with a faster update rate ( more than 2x the real time) the uav gets unstable and eventually falls.

tbanus, 2022-03-01:
I am having a similar issue. I tried it with both Lee Controller and the PID controller I implemented. Both causes issues. My observation is that when the RAM usage reaches %99, the drone falls. However, I haven't tested this. I also suspect the ros time acts differently under these conditions since it messes with the derivative (dt) term of the PID controller. Has anyone reached a solution for this?

Shono1, 2024-07-16:
I ran into this problem as well. Initially, I was keeping with the Gazebo default step size of 0.01 and just increasing the update rate. This caused instability. I fixed my issue by decreasing the step size as well. I did this because I noticed that 0.01 was the max step size for the controller I was using. I figured that working at the max step size was probably a bad idea anyway, as if even one frame was dropped the drone could become unstable. After changing the step size down to 0.005 and upping the update rate to 10 000 hz, I was able to get stable performance (indistinguishable from just a sped up 1x speed) at 5x realtime speed.