ethz-asl/rotors_simulator #699: install on ubuntu 20.04 and ROS noetic

mengchaoheng, 2022-04-24:
need to install ` python3-wstool python3-catkin-tools` instead of ` python-wstool python-catkin-tools`

```
$ sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu `lsb_release -sc` main" > /etc/apt/sources.list.d/ros-latest.list'
$ wget http://packages.ros.org/ros.key -O - | sudo apt-key add -
$ sudo apt-get update
$ sudo apt-get install ros-kinetic-desktop-full ros-kinetic-joy ros-kinetic-octomap-ros ros-kinetic-mavlink python3-wstool python3-catkin-tools protobuf-compiler libgoogle-glog-dev ros-kinetic-control-toolbox ros-kinetic-mavros
$ sudo rosdep init
$ rosdep update
$ source /opt/ros/kinetic/setup.bash
```

mengchaoheng, 2022-04-24:
3.1. If you don't have `rotors_simulator`
   
   ``` 
   $ cd ~/catkin_ws/src
   $ git clone https://github.com/ethz-asl/rotors_simulator.git
   ```

marios-stam, 2022-11-02:
Hello, I followed your instructions but I get the following error:
[image]

Could you please help me?

mengchaoheng, 2022-11-02:
you need python 3？

marios-stam, 2022-11-02:
yes, It seems like that

giabattag, 2023-01-29:
Hello, I also followed the instructions and I got this error:

`------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ecn/ros1/install:/opt/ros/noetic
Workspace:                   /home/ecn/ros
------------------------------------------------------------------
Build Space:        [exists] /home/ecn/ros/build
Devel Space:        [exists] /home/ecn/ros/devel
Install Space:      [unused] /home/ecn/ros/install
Log Space:          [exists] /home/ecn/ros/logs
Source Space:       [exists] /home/ecn/ros/src
DESTDIR:            [unused] None
------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
------------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------------
[build] Found 17 packages in 0.0 seconds.                                                                                                                          
[build] Package table is up to date.                                                                                                                               
Starting  >>> mav_msgs                                                                                                                                             
Starting  >>> mav_state_machine_msgs                                                                                                                               
Finished  <<< mav_state_machine_msgs                [ 1.1 seconds ]                                                                                                
Starting  >>> mav_system_msgs                                                                                                                                      
Finished  <<< mav_system_msgs                       [ 1.3 seconds ]                                                                                                
Starting  >>> mavlink                                                                                                                                              
Finished  <<< mav_msgs                              [ 2.8 seconds ]                                                                                                
Starting  >>> mav_planning_msgs                                                                                                                                    
___________________________________________________________________________________________________________________________________________________________________
Errors     << mavlink:make /home/ecn/ros/logs/mavlink/build.make.003.log                                                                                           
Traceback (most recent call last):
  File "/home/ecn/ros/src/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/ecn/ros/src/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ModuleNotFoundError: No module named 'future'
make[2]: *** [CMakeFiles/icarous.xml-v1.0.dir/build.make:63: include/v1.0/icarous/icarous.h] Error 1
make[1]: *** [CMakeFiles/Makefile2:130: CMakeFiles/icarous.xml-v1.0.dir/all] Error 2
make: *** [Makefile:118: all] Error 2
cd /home/ecn/ros/build/mavlink; catkin build --get-env mavlink | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

...................................................................................................................................................................
Failed     << mavlink:make                          [ Exited with code 2 ]                                                                                         
Failed    <<< mavlink                               [ 0.8 seconds ]                                                                                                
Abandoned <<< rotors_comm                           [ Unrelated job failed ]                                                                                       
Abandoned <<< rotors_control                        [ Unrelated job failed ]                                                                                       
Abandoned <<< rotors_description                    [ Unrelated job failed ]                                                                                       
Abandoned <<< rotors_evaluation                     [ Unrelated job failed ]                                                                                       
Abandoned <<< rotors_hil_interface                  [ Unrelated job failed ]                                                                                       
Abandoned <<< rotors_joy_interface                  [ Unrelated job failed ]                                                                                       
[... cut ...]
[build]   Warnings:  None.                                                                                                                                         
[build]   Abandoned: 12 packages were abandoned.                                                                                                                   
[build]   Failed:    1 packages failed.                                                                                                                            
[build] Runtime: 5.2 seconds total.     `

What is the problem here? Could you please help me?

Jaeyoung-Lim, 2023-01-29:
@mengchaoheng You are missing a python dependency: 
```
ModuleNotFoundError: No module named 'future'
```

giabattag, 2023-01-29:
Thanks a lot, yeah I did not know that "future" was an actual python pip module, i confused it with __future__

> @mengchaoheng You are missing a python dependency:
> 
> ```
> ModuleNotFoundError: No module named 'future'
> ```

Leo-Chao, 2023-05-07:
CMake Error: File /home/liaochao/catkin_ws_CICRSIM/src/mavlink/config.h.in does not exist.
CMake Error at /home/liaochao/catkin_ws_CICRSIM/src/mavlink/CMakeLists.txt:68 (configure_file):
  configure_file Problem configuring file

the problem  how to solve?missing file

gorghino, 2023-05-16:
Hi @Leo-Chao 
I fixed it using the mavlink's git: https://mavlink.io/en/getting_started/installation.html

Leo-Chao, 2023-05-30:
> Hi @Leo-Chao I fixed it using the mavlink's git: https://mavlink.io/en/getting_started/installation.html

yes, I got it before u share it with me, but i still thanks.

kashifzr, 2023-09-15:
rotor_gazebo_plugins failing while catkin build is used, I am unable to find problem in CMakeList.txt specific to rotor_gazebo_plugins

Failed     << rotors_gazebo_plugins:make            [ Exited with code 2 ]                                                                                                                                
Failed    <<< rotors_gazebo_plugins                 [ 22.5 seconds ]                                                                                                                                      
Abandoned <<< rotors_gazebo                         [ Unrelated job failed ]                                                                                                                              
Abandoned <<< rotors_simulator                      [ Unrelated job failed ]                                                                                                                              
[build] Summary: 14 of 17 packages succeeded.                                                                                                                                                             
[build]   Ignored:   None.                                                                                                                                                                                
[build]   Warnings:  15 packages succeeded with warnings.                                                                                                                                                 
[build]   Abandoned: 2 packages were abandoned.                                                                                                                                                           
[build]   Failed:    1 packages failed.                                                                                                                                                                   
[build] Runtime: 29.3 seconds total.                                                                                                                                                                      
[build] Note: Workspace packages have changed, please re-source setup files to use them.

OS = Focal and ROS=Noetic

yifeichen2024, 2024-07-15:
> yes, It seems like that

Hi, I have the same problem, but i have  python3 already. how did you solve this?