ethz-asl/rotors_simulator #710: camera_depth_optical_center_link does not exist

mrajput12, 2022-06-20:
Hi everyone 
This is first time that i am writing here a problem with the hope that i can find solution.i am new  at using rviz and  simulation , i have just cloned  repo from https://github.com/ethz-asl/rotors_simulator .  after  completing installation . i run rviz  .i can see fixed frame  in rviz  when i add point_cloud2 .it shows error . any idea where i am wrong . 
[image]

laurenbramblett, 2023-05-12:
Hello! Were you able to solve this issue? I am having the same problem

laurenbramblett, 2023-08-03:
For anyone having this problem... I fixed it with a transform, but if there is a better solution, I would be interested!
<node pkg="tf2_ros" type="static_transform_publisher" name="link_vi_sensor_broadcaster" args="0 0 0 1.57 3.14 1.47 firefly/base_link camera_depth_optical_center_link" />