ethz-asl/rotors_simulator #714: catkin build -> rotors_gazebo_plugins:make error

Ayazzia01, 2022-10-03:
Ubuntu 20.04
protoc --version == libprotoc 3.6.1

[image]

zhiyu2020, 2022-10-31:
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/rotors_gazebo_noisydepth_plugin.dir/build.make:63: CMakeFiles/rotors_gazebo_noisydepth_plugin.dir/src/gazebo_noisydepth_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1954: CMakeFiles/rotors_gazebo_noisydepth_plugin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/rotors_gazebo_motor_model.dir/build.make:63: CMakeFiles/rotors_gazebo_motor_model.dir/src/gazebo_motor_model.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2251: CMakeFiles/rotors_gazebo_motor_model.dir/all] Error 2
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/rotors_gazebo_ros_interface_plugin.dir/build.make:63: CMakeFiles/rotors_gazebo_ros_interface_plugin.dir/src/gazebo_ros_interface_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:523: CMakeFiles/rotors_gazebo_ros_interface_plugin.dir/all] Error 2

njpzn, 2022-11-25:
I also have this problem,can you tell me how to fix it, Thanks

Faivex, 2024-07-11:
Check my answer [here](https://github.com/gsilano/CrazyS/issues/105#issuecomment-2186404250),
hope that helps.