ethz-asl/rotors_simulator #722: rotor_gazebo_plugins

kashifzr, 2023-09-19:
OS = 20.04 Focal and ROS= Noetic
When I search template argument deduction/substitution failed specific to boost, i found it specific to code compiler, On my computer my gcc compiler is 
<pre>gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
</pre>

my catkin build failed in compile rotor_gazebo_plugin and abandoning rotors_gazebo, When i compile other plugins it make no error

<pre>/usr/include/c++/9/tuple:1341:5: note:   template argument deduction/substitution failed:
In file included from /opt/ros/noetic/include/ros/message.h:38,
                 from /opt/ros/noetic/include/ros/publisher.h:33,
                 from /opt/ros/noetic/include/ros/node_handle.h:32,
                 from /opt/ros/noetic/include/dynamic_reconfigure/server.h:50,
                 from /home/rob-kkj/catkin_ws_rotor/src/rotors_simulator/rotors_gazebo_plugins/include/rotors_gazebo_plugins/gazebo_noisydepth_plugin.h:31,
                 from /home/rob-kkj/catkin_ws_rotor/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_noisydepth_plugin.cpp:24:
/usr/include/boost/array.hpp:440:7: note: candidate: ‘template&lt;long unsigned int Idx, class T, long unsigned int N&gt; T&amp; std::get(boost::array&lt;T, N&gt;&amp;)’
  440 |    T &amp;get(boost::array&lt;T,N&gt; &amp;arr) BOOST_NOEXCEPT {
      |       ^~~
/usr/include/boost/array.hpp:440:7: note:   template argument deduction/substitution failed:
/usr/include/boost/array.hpp:446:13: note: candidate: ‘template&lt;long unsigned int Idx, class T, long unsigned int N&gt; const T&amp; std::get(const boost::array&lt;T, N&gt;&amp;)’
  446 |    const T &amp;get(const boost::array&lt;T,N&gt; &amp;arr) BOOST_NOEXCEPT {
      |             ^~~
/usr/include/boost/array.hpp:446:13: note:   template argument deduction/substitution failed:
make[2]: *** [CMakeFiles/gazebo_geotagged_images_plugin.dir/build.make:76: CMakeFiles/gazebo_geotagged_images_plugin.dir/src/external/gazebo_geotagged_images_plugin.cpp.o] Error 1
make[2]: *** [CMakeFiles/rotors_gazebo_noisydepth_plugin.dir/build.make:76: CMakeFiles/rotors_gazebo_noisydepth_plugin.dir/src/gazebo_noisydepth_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3618: CMakeFiles/gazebo_geotagged_images_plugin.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:4154: CMakeFiles/rotors_gazebo_noisydepth_plugin.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
</pre>

..........................................................................................................................................................................................................
Failed     << rotors_gazebo_plugins:make            [ Exited with code 2 ]                                                                                                                                
Failed    <<< rotors_gazebo_plugins                 [ 22.3 seconds ]                                                                                                                                      
Abandoned <<< rotors_gazebo                         [ Unrelated job failed ]                                                                                                                              
Abandoned <<< rotors_simulator                      [ Unrelated job failed ]                                                                                                                              
[build] Summary: 13 of 16 packages succeeded.                                                                                                                                                             
[build]   Ignored:   None.                                                                                                                                                                                
[build]   Warnings:  13 packages succeeded with warnings.                                                                                                                                                 
[build]   Abandoned: 2 packages were abandoned.                                                                                                                                                           
[build]   Failed:    1 packages failed.                                                                                                                                                                   
[build] Runtime: 25.0 seconds total.

closed 2023-09-25 by kashifzr

anish-bhattacharya, 2023-10-16:
@kashifzr were you able to resolve this? I'm also using Ubuntu 20 / ROS Noetic / Gazebo 11 ; I imagine the build errors are because the necessary libignition version is different with the _supported_ ROS Melodic & Gazebo 9.

kashifzr, 2023-10-16:
@anish-bhattacharya I am using this simulator with out any issue after changing Ubuntu but Ubuntu version is still same, When i was trying to build rotor_gazebo_plugin its creates issues and i did not find any help.